zephyrproject-rtos/zephyr #11087: [Coverity CID :189519]Error handling issues in /samples/net/sockets/echo/src/socket_echo.c

mandarcthorat1, 2018-11-05:
Static code scan issues seen in File: /samples/net/sockets/echo/src/socket_echo.c
 Category: Error handling issues
 Function: main
 Component: Samples
 CID: 189519
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nniranjhana, 2018-11-06:
This looks intentional to me. I don't see why the arguments (SOCKET) can't be negative for `bind` and `recv` calls.

I think we can mark so in Coverity and close this. Same applies to #11086

nashif, 2019-03-05:
@pfalcon please take a look at this issue and close ASAP

tedd-an, 2019-03-14:
@galak or @nashif 
This can be closed.

rveerama1, 2019-03-22:
Fix already merged, so closed.

closed 2019-03-22 by rveerama1