zephyrproject-rtos/zephyr #11089: [Coverity CID :189517]Control flow issues in /drivers/sensor/fxos8700/fxos8700.c

mandarcthorat1, 2018-11-05:
Static code scan issues seen in File: /drivers/sensor/fxos8700/fxos8700.c
 Category: Control flow issues
 Function: fxos8700_init
 Component: Drivers
 CID: 189517
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

himanshujha199640, 2018-11-06:
False postive as the switch case branch skims through various
WHOAMI register to check for various supported devices.

And @ceolin I think it doesn't violate MISRA-C. Isn't it ?
https://github.com/zephyrproject-rtos/zephyr/issues/9893

subbu147, 2018-11-06:
It looks like a valid waring.
It violates the MISRA-C
switch (x) {
   case 'a':  /* allowed */
   case 'b':
      do_something();
      break;
  **case 'c':
      do_something_else(); /*not allowed */
  case 'd':
      break;**
  default:
      break;
}