zephyrproject-rtos/zephyr #11091: [Coverity CID :189515]Incorrect expression in /drivers/usb/device/usb_dc_stm32.c

mandarcthorat1, 2018-11-05:
Static code scan issues seen in File: /drivers/usb/device/usb_dc_stm32.c
 Category: Incorrect expression
 Function: usb_dc_stm32_get_maximum_speed
 Component: Drivers
 CID: 189515
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aurel32, 2018-11-05:
The Coverity analysis consider the data (here the maximum-speed) value to be a constant. In practice it is not the case, different SoC have different values.

So it looks like a false positive to me.

ydamigos, 2018-11-20:
@galak How can I access Coverity's report? Should I request "Organization access" permission?

galak, 2018-11-21:
> @galak How can I access Coverity's report? Should I request "Organization access" permission?

yes, you need to request access.

galak, 2018-11-21:
Agree with @aurel32 assessment, this is intentional because of Device Tree.

closed 2018-11-21 by galak