zephyrproject-rtos/zephyr #11095: [Coverity CID :189511]Code maintainability issues in /subsys/settings/src/settings_fcb.c

mandarcthorat1, 2018-11-05:
Static code scan issues seen in File: /subsys/settings/src/settings_fcb.c
 Category: Code maintainability issues
 Function: settings_fcb_compress
 Component: Other
 CID: 189511
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

himanshujha199640, 2018-12-07:
The code in context is:
```
211                 rc = fcb_append_finish(&cf->cf_fcb, &loc2.loc);
212                 __ASSERT(rc == 0, "Failed to finish fcb_append.\n");
213         }
214         rc = fcb_rotate(&cf->cf_fcb);
215 
216         __ASSERT(rc == 0, "Failed to fcb rotate.\n");
```

And coverity says:
```
CID 189511 (#1 of 1): Unused value (UNUSED_VALUE)
returned_value: Assigning value from fcb_append_finish(&cf->cf_fcb, &loc2.loc) to rc here, but that stored value is overwritten before it can be used.
```

which means coverity doesn't understand the purpose of `__ASSERT` and the returrn value is actually **used** there. So, clearly it is another class of false postives I have observed while skimming through
other coverity issues.

So, please mark it as `False positive` in the Coverity triage, admins!
I don't have the rights to do it myself and closing this issue wouldn't be a nice
solution because next time again coverity will spit out the same issue.

nvlsianpu, 2018-12-10:
because of #9521 this issue will be not valid anymore.

himanshujha199640, 2018-12-10:
> because of #9521 this issue will be not valid anymore.

OK. Then we would close this issue once the PR gets merged.
But best to mark it as `False positive`.

closed 2018-12-12 by nvlsianpu