zephyrproject-rtos/zephyr #11097: [Coverity CID :189509]Integer handling issues in /drivers/sensor/ms5837/ms5837.c

mandarcthorat1, 2018-11-05:
Static code scan issues seen in File: /drivers/sensor/ms5837/ms5837.c
 Category: Integer handling issues
 Function: ms5837_compensate
 Component: Drivers
 CID: 189509
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

vanwinkeljan, 2018-11-05:
requested access to zephyr coverity project for further investigation

vanwinkeljan, 2018-11-05:
Coverity reported possible overflow at line 74 in [ms5837.c:](https://github.com/zephyrproject-rtos/zephyr/blob/c3d75488d408d618cd823be162770efa627684ce/drivers/sensor/ms5837/ms5837.c#L74)
```
	temp_sq = (data->temperature - 2000) * (data->temperature - 2000);
```

But as per MS5837 datasheet the maximum value of ```data->temperature``` is 8500 and the minimum value is -4000 as such no overflow can occur.

vanwinkeljan, 2018-11-05:
@agross-linaro, @ceolin could your mark the coverity issue as intentional?

ceolin, 2018-12-12:
@vanwinkeljan done, thanks !

closed 2018-12-12 by ceolin