zephyrproject-rtos/zephyr #11109: DeviceTree metadata should be available to CMake

SebastianBoe, 2018-11-05:
CMake parses the Kconfig output for CONFIG_* symbols and adds them to the CMake namespace.

It does the same for DT, but because DT doesn't have a namespace (https://github.com/zephyrproject-rtos/zephyr/issues/7213) it filters on DT symbols with CONFIG_* and loses most of the DT information.

For DeviceTree to be useful, it's information needs to be accessible from CMake, just like Kconfig is.

This feature is blocked by #7213, as one cannot introduce DT symbols into the CMake namespace without any namespace scoping rules that prevent aliasing issues.

erwango, 2018-11-06:
@SebastianBoe , in order to try to unblock this ticket.
Can you clarify "namespace scoping rules that prevent aliasing issues" (I think I get it but I prefer to be sure to avoid any misunderstanding) and what in #7213 is blocking this?

SebastianBoe, 2018-11-06:
In any namespace (set of symbols), you need some way to prevent aliasing issues.

So for instance.

In the CMake namespace (set of CMake variable names), symbols named CONFIG_* are reserved for Kconfig. It is not permitted to create a CMake variable that is prefixed with CONFIG_.

Similairly, in the the C namespace symbols named CONFIG_* are reserved for Kconfig. It is not permitted to #define CONFIG_FOO, or to have a variable prefixed with CONFIG_. (This rule is violated at times because we have no way of enforcing it).

For device tree symbols to be available to C code and to CMake build scripts, they also need to have some prefix to avoid aliasing issues.

erwango, 2018-11-06:
So is that all about defining a DTS_ (or DT_) prefix for device tree generated symbols?

SebastianBoe, 2018-11-06:
Yes.

mnkp, 2018-12-12:
Just a quick comment: #7213 was largely addressed by #11180. Almost all of the DeviceTree symbols have now DT_ prefix. As such, this issue is no longer blocked.

PR #12046 (2018-12-13): Import dt into cmake namespace
CMake parses the Kconfig output for 'CONFIG_*' symbols and adds them
to the CMake namespace.

It does the same for DT, but has only been including 'CONFIG_' symbols
and been ignoring the 'DT_' symbols.

For DeviceTree to be useful, it's information needs to be accessible
from CMake, just like Kconfig is.

To this end we also introduce the 'DT_' symbols into the CMake
namespace.

This resolves #11109
commit 2192302c (2018-12-13): cmake: DT: Introduce 'DT_' symbols into the CMake namespace

CMake parses the Kconfig output for 'CONFIG_*' symbols and adds them
to the CMake namespace.

It does the same for DT, but has only been including 'CONFIG_' symbols
and been ignoring the 'DT_' symbols.

For DeviceTree to be useful, it's information needs to be accessible
from CMake, just like Kconfig is.

To this end we also introduce the 'DT_' symbols into the CMake
namespace.

This resolves #11109
closed 2018-12-13 by galak