zephyrproject-rtos/zephyr #11111: Shippable shows too little for documentation errors

jukkar, 2018-11-05:
I got following error from shippable
```
doc/_build/rst/doc/api/networking.rst:48: WARNING: Duplicate declaration.
```
But that is not very useful for anything. I then tried to replicate this by running `make htmldocs` manually, and then the real issue was shown.
```
include/net/net_if.h:1462: warning: argument 'iface' of command @param is not found in the argument list of net_if_ipv4_select_src_addr(struct net_if *dst_iface, struct in_addr *dst)
include/net/net_if.h:1472: warning: The following parameters of net_if_ipv4_select_src_addr(struct net_if *dst_iface, struct in_addr *dst) are not documented:
  parameter 'dst_iface'
```

Would it be possible to get the actual error message already to shippable logs?

I am asking this because building docs is a very heavy operation and takes long time. The actual error is still there in the log file when shippable builds the beast, it is just not showing the culprit.

nashif, 2018-11-05:
you have a link to the shippable build?

jukkar, 2018-11-05:
https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/26163/5/tests

jukkar, 2018-11-05:
Other more recent build which failed
https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/26183/5/tests

nashif, 2018-11-05:
@dbkinder any idea?

dbkinder, 2018-11-05:
@nashif the "duplicate declaration" warnings (with no mention of what is duplicated) are from doxygen/breathe and are normally filtered by our ".known_issues" filter.  I see we're using doxygen version 1.8.11 in the CI system and the reporting of these duplicate declarations seems to change with doxygen versions.  I've added a .known_issues filter to remove these.

But the differences in the reported errors you mention in this issue report are not caused by this.  The duplicate declaration warnings have nothing to do with the "argument 'iface' of command @param is not found" warnings.  When I look at the shippable console output you link to, only the duplicate declaration warnings are there, so the "argument 'iface'" warnings are from something else.

dbkinder, 2018-11-06:
@jukkar BTW, I can build docs on my Linux box with a clean build folder (``rm -rf doc/_build``) in just under 7 minutes. If I run it a second time with only one or two .rst files changed, it takes only 1m30s (because of the optimizations built into the build system).  Are you finding it's taking much longer that that for you, and that why you call it a "very heavy operation"?

closed 2018-11-06 by nashif

jukkar, 2018-11-06:
@dbkinder for my machine the document generation takes 30 min (Core-i7 16gb), which is quite long imho. Thanks for fixing the issue!

dbkinder, 2018-11-06:
@jukkar Recent changes to the doc/CMakeLists.txt processing may be the culprit here.  Previously, the doc build created it's output within the doc/ folder structure, mostly within the doc/_build folders, but some files were created outside of this and we used .gitignore to keep Git happy.  Because of the way the current doc build works, it copies .rst files into a build folder before starting the doxygen and sphinx process to build the HTML.  Alas, if you have the "old" folders around in your doc/ folder, it copies them too and tries to create Cmake dependencies for some 18,000 files instead of only the 1000 or so it should.  This causes Cmake to take a l-o-n-g time before it even gets to the doxygen/sphinx step.

So, the fix is to remove the old output folders that (probably) still exist in your doc/ folder, specifically, within the ``doc/`` folder, remove:
```
rm -rf _build/*
rm -rf html
rm -rf xml
rm -rf doxygen
rm -rf latex
rm -rf reference/kconfig/*.rst
rm -rf samples
rm -rf boards
```
This used to be done in the doc/Makefile (``make clean`` but didn't get into the cmake processing).
See if that reduces your doc build time back down to < 7 minutes.

jukkar, 2018-11-07:
> See if that reduces your doc build time back down to < 7 minutes.

That did the trick, doc build is now under 7 minutes in my machine. Thanks for the help!

PR #11124 (2018-11-06): doc: filter expected dup decl warnings
The API documentation for networking.rst can throw a doxygen warning
depending on the versions of sphinx/breathe being used.

Fixes: #11111
commit ae4d7612 (2018-11-06): doc: filter expected dup decl warnings

The API documentation for networking.rst can throw a doxygen warning
depending on the versions of sphinx/breathe being used.

Fixes: #11111