zephyrproject-rtos/zephyr #11121: dts: cannot set child of parent with multiple compatible-s

kzyapkov, 2018-11-05:
Trying to add a modem as child node of a serial controller on STM32L4 results in an error in `extract_dts_includes.py`

```
Traceback (most recent call last):
  File ".../zephyr/scripts/dts/extract_dts_includes.py", line 338, in extract_property
    yaml[reduced[parent_address]['props']['compatible']]
TypeError: unhashable type: 'list'

During handling of the above exception, another exception occurred:

...

Exception: /soc/serial@40008000/bc95 defines parent /soc/serial@40008000 as bus master but /soc/serial@40008000 not configured as bus master in yaml description
```

This is because UART/USARTs on STM32 have a list of compatible strings

    compatible = "st,stm32-usart", "st,stm32-uart";

This patch fixes it for me, but is probably not the correct way to handle the situation

```
diff --git a/scripts/dts/extract_dts_includes.py b/scripts/dts/extract_dts_includes.py
index 9b25c46d22..1468d79de3 100755
--- a/scripts/dts/extract_dts_includes.py
+++ b/scripts/dts/extract_dts_includes.py
@@ -334,8 +334,10 @@ def extract_property(node_compat, yaml, node_address, prop, prop_val, names):
 
             #check parent has matching child bus value
             try:
-                parent_yaml = \
-                    yaml[reduced[parent_address]['props']['compatible']]
+                parent_compat = reduced[parent_address]['props']['compatible']
+                if type(parent_compat) is list:
+                    parent_compat = parent_compat[0]
+                parent_yaml = yaml[parent_compat]
                 parent_bus = parent_yaml['child']['bus']
             except (KeyError, TypeError) as e:
                 raise Exception(str(node_address) + " defines parent " +
```

erwango, 2018-11-06:
@kzyapkov 

Can you try the following?
```
diff --git a/scripts/dts/extract_dts_includes.py b/scripts/dts/extract_dts_includes.py
index 9b25c46..e2db81e 100755
--- a/scripts/dts/extract_dts_includes.py
+++ b/scripts/dts/extract_dts_includes.py
@@ -334,8 +334,8 @@ def extract_property(node_compat, yaml, node_address, prop, prop_val, names):
 
             #check parent has matching child bus value
             try:
-                parent_yaml = \
-                    yaml[reduced[parent_address]['props']['compatible']]
+                parent_compat = get_compat(parent_address)
+                parent_yaml = yaml[parent_compat]
                 parent_bus = parent_yaml['child']['bus']
             except (KeyError, TypeError) as e:
                 raise Exception(str(node_address) + " defines parent " +
-- 
```

kzyapkov, 2018-11-07:
@erwango your patch works and looks like the correct way to handle the situation.

PR #11230 (2018-11-10): scripts/dts: extract_dts_includes: fix multi-compat bus parent
Case when bus parent has more than one compatible was not treated
correctly. Use get_compat() method which returns first compat
in case several compats are available

Fixes #11121
commit eaac6d3c (2018-11-10): scripts/dts: extract_dts_includes: fix multi-compat bus parent

Case when bus parent has more than one compatible was not treated
correctly. Use get_compat() method which returns first compat
in case several compats are available

Fixes #11121
closed 2018-11-10 by galak