zephyrproject-rtos/zephyr #11135: subsys/net/lib/: WARNING: Unsigned expression compared with zero

himanshujha199640, 2018-11-06:
Found using Coccinelle.
https://github.com/zephyrproject-rtos/zephyr/pull/11132
```
himanshu@himanshu-Vostro-3559:~/zephyr$ ./scripts/coccicheck --mode=report --cocci=scripts/coccinelle/unsigned_lesser_than_zero.cocci --verbose=1 subsys/net/

Please check for false positives in the output before submitting a patch.
When using "patch" mode, carefully review the patch before submitting it.

Processing unsigned_lesser_than_zero.cocci
with option(s) ""

Message example to submit a patch:
 Unsigned expressions cannot be lesser than zero. Presence of
 comparisons 'unsigned (<|<=) 0' often indicates a bug,
 usually wrong type of variable.

 The semantic patch that makes this report is available
 in scripts/coccinelle/unsigned_lesser_than_zero.cocci.

 More information about semantic patching is available at
 http://coccinelle.lip6.fr/

Running (4 in parallel): /usr/local/bin/spatch -D report --no-show-diff --very-quiet --cocci-file scripts/coccinelle/unsigned_lesser_than_zero.cocci --patch /home/himanshu/zephyr --dir subsys/net/ --jobs 4 --chunksize 1
subsys/net/lib/mqtt/mqtt_pkt.c:638:5-10: WARNING: Unsigned expression compared with zero.
subsys/net/lib/mqtt/mqtt_pkt.c:793:5-13: WARNING: Unsigned expression compared with zero.
subsys/net/lib/lwm2m/lwm2m_obj_device.c:172:5-16: WARNING: Unsigned expression compared with zero.
```

jukkar, 2018-11-07:
We are deprecating the current mqtt implementation soon so those warnings can be ignored right now.

@mike-scott Can you check the lwm2m warning?

mike-scott, 2018-11-07:
@jukkar I'll check into it

closed 2019-01-30 by jukkar