zephyrproject-rtos/zephyr #11136: drivers/spi: WARNING: Unsigned expression compared with zero

himanshujha199640, 2018-11-06:
Found using Coccinelle.
https://github.com/zephyrproject-rtos/zephyr/pull/11132
```
himanshu@himanshu-Vostro-3559:~/zephyr$ ./scripts/coccicheck --mode=report --cocci=scripts/coccinelle/unsigned_lesser_than_zero.cocci --verbose=1 drivers/spi/

Please check for false positives in the output before submitting a patch.
When using "patch" mode, carefully review the patch before submitting it.

Processing unsigned_lesser_than_zero.cocci
with option(s) ""

Message example to submit a patch:
 Unsigned expressions cannot be lesser than zero. Presence of
 comparisons 'unsigned (<|<=) 0' often indicates a bug,
 usually wrong type of variable.

 The semantic patch that makes this report is available
 in scripts/coccinelle/unsigned_lesser_than_zero.cocci.

 More information about semantic patching is available at
 http://coccinelle.lip6.fr/

Running (4 in parallel): /usr/local/bin/spatch -D report --no-show-diff --very-quiet --cocci-file scripts/coccinelle/unsigned_lesser_than_zero.cocci --patch /home/himanshu/zephyr --dir drivers/spi/ --jobs 4 --chunksize 1
drivers/spi/spi_sam0.c:222:5-8: WARNING: Unsigned expression compared with zero.
drivers/spi/spi_sam.c:215:5-8: WARNING: Unsigned expression compared with zero.
```

henrikbrixandersen, 2018-11-20:
Do we have the commit hash on which the static code analysis was run? File `drivers/spi/spi_sam0.c:222` is `spi_sam0_fast_txrx()` in current master (3abb8a852155cf87bdbbbf2c7e334162d2c8c2d8).

himanshujha199640, 2018-11-20:
> Do we have the commit hash on which the static code analysis was run? File `drivers/spi/spi_sam0.c:222` is `spi_sam0_fast_txrx()` in current master ([3abb8a8](https://github.com/zephyrproject-rtos/zephyr/commit/3abb8a852155cf87bdbbbf2c7e334162d2c8c2d8)).

What do you mean ?
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/spi/spi_sam0.c#L231

PR #11695 (2018-11-27): drivers: spi: sam{,0}: avoid unsigned comparision with less than zero
Avoid comparison of size_t with less than zero.
This fixes #11136.
commit 4c533d05 (2018-11-27): drivers: spi: sam: avoid unsigned comparision with less than zero

Avoid comparison of size_t with less than zero.

This fixes #11136.
commit b52b79d1 (2018-11-27): drivers: spi: sam0: avoid unsigned comparision with less than zero

Avoid comparison of size_t with less than zero.

This fixes #11136.
closed 2018-11-27 by andrewboie