zephyrproject-rtos/zephyr #11171: usb: nrf: usb_nrfx is in endless loop when USB cable in not connected

finikorg, 2018-11-07:
When USB cable is not connected to the `reel_board`'s `nRF USB` connector board issues endless events:

```
...
[00:06:40.146,301] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.149,414] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
[00:06:40.149,414] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.169,036] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.232,116] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.236,267] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.237,304] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.240,417] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
[00:06:40.240,447] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.259,277] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.260,314] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.263,427] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
[00:06:40.263,427] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.283,416] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.346,527] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.350,677] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.351,715] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.354,827] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
[00:06:40.354,827] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.374,206] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.375,244] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.378,356] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
[00:06:40.378,356] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.397,521] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.460,632] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.464,782] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.465,820] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.468,933] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
[00:06:40.468,933] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.487,548] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.488,586] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.491,668] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
[00:06:40.491,699] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.511,901] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.575,012] <dbg> usb_nrfx.usbd_handle_state_change: USB Removed
[00:06:40.579,132] <dbg> usb_nrfx.usbd_handle_state_change: USB detected
[00:06:40.580,200] <dbg> usb_nrfx.usbd_handle_state_change: USB Powered
[00:06:40.583,282] <dbg> usb_nrfx.usbd_event_handler: USBD Reset.
...
```

pawelzadrozniak, 2018-11-13:
I was able to recreate it.
The same binary works fine on nRF52DK. Could it be the board issue?

carlescufi, 2018-11-28:
@jfischer-phytec-iot this does not happen with the nRF52840 DK. Could it be hardware related?

jfischer-phytec-iot, 2018-11-28:
I can reproduce it with `CONFIG_USB_DRIVER_LOG_LEVEL_DBG=y`. It seems that the VBUS_nRF floats and rises the interrupt mentioned above.

jfischer-phytec-iot, 2018-11-28:
The reverse leakage current over D6 charges the capacitor C61 until nRF USB detects "USB Powered", this in turn discharges the capacitor C61 and the whole thing repeats itself. This can only be fixed by desoldering  of the D6, but I suggest to just plug in the cable. Will be fixed in next hardware revision.

carlescufi, 2018-11-28:
Hardware issue

closed 2018-11-28 by carlescufi