zephyrproject-rtos/zephyr #11176: Support for particle boards

Lindenk, 2018-11-07:
Support for Particle's boards would be very helpful. This includes (in the order if my prefrence):

- Xenon
- Boron (although this requires GSM support)
- Argon

Also, support for the particle programmer/debugger would be nice too (if it isn't already).

pabigot, 2019-01-08:
The debugger (flasher) is a standard daplink that works with pyocd.  It's the default board runner in the PR.

I have no idea how those documents were made visible.  They're from one of the original pull requests.

sbrichardson, 2019-01-09:
Google indexed them, that’s how I landed on it, although it confused me when refreshing on the current site (where’d they go?), until I realized after reading the PRs :).

Do you think it’s because of its feather pin out/Arduino compatibility that it just works? Still haven’t tested, expect to later this week.

pabigot, 2019-01-12:
closed by #12195 ; thanks @nashif

closed 2019-01-12 by pabigot