zephyrproject-rtos/zephyr #11187: cmake/toolchain/clang: selecting clang doesn't select the correct compiler file

sigvartmh, 2018-11-07:
When setting `ZEPHYR_TOOLCHAIN_VARIANT=clang` and try to compile an example you'll get 
`-- Performing Test toolchain_is_ok - Failed`
This is due to this error:

```
FAILED: cmTC_80be7 
: && /usr/bin/clang --target=i686-pc-none-elf -Dtoolchain_is_ok   CMakeFiles/cmTC_80be7.dir/src.c.obj  -o cmTC_80be7   && :
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-pc-linux-gnu/8.2.1/../../../libgcc_s.so.1 when searching for libgcc_s.so.1
/usr/bin/ld: skipping incompatible /usr/lib/libgcc_s.so.1 when searching for libgcc_s.so.1
/usr/bin/ld: cannot find libgcc_s.so.1
collect2: error: ld returned 1 exit status
clang-7: error: linker (via gcc) command failed with exit code 1 (use -v to see invocation)
```
This error is caused by the following line: https://github.com/zephyrproject-rtos/zephyr/blob/58781d5782fc7644dafd44d309ef485eee3f88d9/cmake/toolchain.cmake#L71 

due to it trying to compile something called `compiler/clang.cmake` however there is a file called `llvm.cmake` if you change the name of it to `clang.cmake` at least the cmake generation is successful.

Haven't been able to build anything so I wonder a bit about the intention of this file and how clang is supposed to be invoked.

carlescufi, 2018-12-07:
CC @mped-oticon

SebastianBoe, 2018-12-07:
I think I'm going to have to insist that someone else fixes this issue. If I can't make something work, I can't fix it's bugs either.

PR #11992 (2018-12-13): Clang fixes
This PR contains misc. clang fixes and cleanups.

Including;
fixes #11187
resolves https://github.com/zephyrproject-rtos/zephyr/pull/11810#discussion_r239618154
commit bb278b68 (2018-12-13): cmake: toolchain: Change the names of 'clang' and 'llvm'

The 'llvm' and 'clang' build scripts have been named strangely. It is
more natural for 'clang' to be the compiler, and 'llvm' to be the
toolchain.

This commit rectifies this by renaming the files.

This also fixes #11187
closed 2018-12-13 by galak