zephyrproject-rtos/zephyr #11198: Bluetooth Mesh: device receives it's own sent messages when CONFIG_BT_SETTINGS is enabled

RonAmihai, 2018-11-07:
@jhedberg 

I'm opening this issue further to my previous one, which is also related to Bluetooth Mesh. Maybe the issues are related.
https://github.com/zephyrproject-rtos/zephyr/issues/10952

When CONFIG_BT_SETTINGS is enabled, and messages are sent through publication context, I've noticed that the same device receives it own messages. 

In my situation, the publication context sends message with opcode 0x40, and fails afterwards with debug message "No OpCode 0x00000040 for elem 0" because this opcode isn't bound to any model (there is no point to be bound to opcode which is used to send outgoing messages).

This debug message is printed in function bt_mesh_model_recv (mesh/access.c). 

Disabling CONFIG_BT_SETTINGS solves the above issue, somehow, even though outgoing messages opcodes (as part of the message's buffer) shouldn't be stored anyway

jhedberg, 2018-11-08:
What destination address are you using? Are you sure the local node doesn't subscribe to it?

RonAmihai, 2018-11-10:
Erasing the device's flash using nrfjprog utility (-e / --eraseall) actually solved this issue for us. 
Thanks for helping anyway.

closed 2018-11-10 by RonAmihai