zephyrproject-rtos/zephyr #11202: recent "shell: shell_uart" commit breaks shell samples on cc3220sf

GAnthony, 2018-11-08:
The commit: 4962618e "shell: shell_uart: add ring_buffers and interrupt support" breaks the samples/subsys/shell/shell_module and samples/net/wifi shell samples on the cc3220sf_launchxl, resulting in no character input.

Reverting that commit makes the samples work again.

The interrupt driven uart tests had previously been working, so there is some assumption being made with this new shell client that isn't being met by the cc32xx_uart.

jarz-nordic, 2018-11-09:
@GAnthony : I tested it with no problems on:
- all Nordic boards
- frdm_k64f
- disco_l475_iot1
and it was working fine.

In my opinion this is problem with serial driver - interrupt API. You can unselect UART interrupt API in menuconfig:
`Device Drivers -> Serial Drivers -> Enable UART Interrupt support`. Now shell will work in polling mode like before this commit.

nordic-krch, 2018-11-09:
Most likely it's error in TI serial driver and i won't be able to help with that.

pfalcon, 2018-11-09:
@GAnthony : Please try samples/subsys/console/echo on you board.

And see https://github.com/zephyrproject-rtos/zephyr/issues/8869 for more cases (but that issue is dedicated to QEMU boards).

GAnthony, 2018-11-09:
The design of the UART interrupt API is problematic.
It expects the driver to "cache" the results of the interrupt status register, to (hopefully at the right time) be read by the client's callback function.  Drivers shouldn't be doing this, as it creates races between the cached value and the actual interrupt status register value at a later time.

Yesterday, I rewrote the driver per the updated uart.h comments (thanks @pfalcon for that), but still get no characters echoed via the shell.

I'll try to adapt this to the shell use case, but I may just revert to polling mode until the new UART API is available.

GAnthony, 2018-11-10:
> You can unselect UART interrupt API in menuconfig:
> Device Drivers -> Serial Drivers -> Enable UART Interrupt support. Now shell will work in polling mode like before this commit.

OK, I did that, and tested that shell does indeed work.  However, now each log message is getting printed twice (with the same time stamp).  And, also, console does not work, since it's interrupt driven only.

Since I need shell more than console right now, I'll do this workaround and file an issue on uart_cc32xx.c to either 1) try to conform it's interrupt handling to the expectations of shell/console; or 2) wait for the new uart API.

GAnthony, 2018-11-10:
I noted that console expects that a call to  `putchar->uart_irq_tx_enable()` will trigger a tx interrupt.  But in fact, the tx interrupt is only triggered after the uart sends a character, so that interrupt actually gets miss

pfalcon, 2018-11-10:
> I noted that console expects that a call to putchar->uart_irq_tx_enable() will trigger a tx interrupt. But in fact, the tx interrupt is only triggered after the uart sends a character, 

So, we've got a problem here. As comments to uart.h should explain (if they don't, d'oh), there're (at least) 2 possible TX interrupts: a) FIFO has some empty space and *ready* to accommodate more chars; b) UART has *completed* sending out chars. Ready/complete terminology might be not ideal, but an improvement over previous "empty" one we had, where it was completely ambiguous whether it was "somewhat empty" or "completely empty". "tx_ready" in particular symmetric with "rx_ready" - when it's "ready", it can do useful work to us - give out, or take in, a character.

Note that "complete" interrupt is not really useful for *data transfer*. Unfortunately, your quote above describes the behavior of "complete" interrupt, not "ready".

pfalcon, 2018-11-10:
> I previously disabled the FIFOs on the cc32xx, to simplify processing

So, I studied both the driver and cc3220 TRM, and here's my analysis and suggestions to try to fix the situation. First of all, I indeed was surprised that FIFO wasn't used in the current driver. But my reading thru the TRM, "disabling FIFO" effectively reducing it to be of the depth 1. All other interrupt processing works in the consistent way whether it's disabled or not. So, indeed, disabling it is indeed a way to make the driver simpler, and let's not re-enable it until we fixed the current situation.

Then, swru465.pdf, page 167:

> EOT
> End of Transmission
This bit determines the behavior of the TXRIS bit in the UARTRIS
register.
0h = The TXRIS bit is set when the transmit FIFO condition specified
in UARTIFLS is met.
1h = The TXRIS bit is set only after all transmitted data, including
stop bits, have cleared the serializer.

So, here we go, while some other chips have separate TX "ready" and "complete" interrupts, cc3220 allows to select one or the other behavior. We need "ready", again, and it appears to be default (0), but for the avoidance of doubt, can you please add explicit configuring of it to uart_cc32xx_init()?

Next, p.172.

> TXRIS
> UART Transmit Raw Interrupt Status
> This bit is cleared by writing a 1 to the TXIC bit in
the UARTICR register or by writing data to the transmit FIFO until it
becomes greater than the trigger level, if the FIFO is enabled, or by
writing a single byte if the FIFO is disabled.

Similarly for RX. So, recap: many chips control acknowledgement of RX/TX implicitly based of FIFO status. Like, if there's free space in TX fifo, you'll get an interrupt. ISR should feed an interrupt. If you feed too little, on return from ISR you'll immediately get another interrupt. Once you fed enough, it'll subside for a while. That's "behavior contract" on which e.g. subsys/console/tty.c is based. And it's not important which depth of FIFO has - 1 or larger, it should work just the same.

So, cc3220 is like that too. But in addition to implicit IRQ control based on FIFO, it also has explicit control, like we can forcibly "acknowledge" (actually reset) RX/TX interrupts. Well, it may be useful at times, but it is **bad** idea to use both implicit and explicit control - given that implicit already works like Zephyr API expects of it, there can be only conflict if mix in explicit control.

So, we now can proceed with identifying actual problem spots in the driver:

1. https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/serial/uart_cc32xx.c#L75
~~~
MAP_UARTIntClear((unsigned long)config->base,
(UART_INT_RX | UART_INT_TX));
~~~

So, what's the natural state of UART interrupts after init? RX didn't receive anything yet, so RX interrupt is off. But TX block is ready to send something, so TX interrupt should be on! Here you forcibly reset it, and that would be the first cause of incorrect behavior. So, please replace `(UART_INT_RX | UART_INT_TX)` with `UART_INT_RX`. And actually, please readback INT status register after that and check that it has TX INT bit set. If not, something else resets it (not hardware, per my reading of the TRM), and we have big problem, and would need to either un-reset it, or emulate the behavior describe above ("TX int is on after UART init") to bootstrap the processing.

2. https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/serial/uart_cc32xx.c#L284
~~~
MAP_UARTIntClear((unsigned long)config->base, intStatus);
~~~

Here you again forcibly "acknowledge" interrupts. There's no need to do that, as explained above, they're acknowledged implicitly based on FIFO state. And explained above, subsys/console/tty (and anything else in general) relies on the fact that if FIFO is not fully fed after one call to ISR, then it will be called again.

Don't remove that line altogether though, because here I talk only about data, RX/TX interrupts. We should not forget about error interrupts. Unfortunately, error condition handling in current API - drivers, their users - leaves much to be desired. So, I suggest to explicitly ack any possible error INTs, to avoid situations like https://github.com/zephyrproject-rtos/zephyr/pull/11039 . 

In other words, that line would become 

~~~
MAP_UARTIntClear((unsigned long)config->base, intStatus & ~(UART_INT_RX | UART_INT_TX));
~~~

(with comment above updated too of course ;-) )

qianfan-Zhao, 2018-11-11:
#11281 it's the same thing as this one

andrewboie, 2018-11-13:
> Well, that right time should be "in ISR, few-CPU-instruction-wise immediately after caching", so possibility of races should be minimized.

Minimized, but nonzero?

nordic-krch, 2018-11-13:
@GAnthony regarding:
>  However, now each log message is getting printed twice (with the same time stamp).

see #11320

jarz-nordic, 2018-11-13:
@GAnthony 
Similar problem we had in Nordic UARTE driver. What I did to have TX interrupt active was to send 0 lenght message during driver initialization.

PR #11277 (2018-11-19): drivers: serial: Update cc32xx uart interrupt handling
The Zephyr console and shell interrupt processing assumes
a TX interrupt is evoked upon first enabling the TX
interrupt via uart_irq_tx_enable.

This was not the case with the cc32xx uart, coming out of
reset, with FIFO's disabled.

The only way found to achieve this behavior is to fill
the fifo with a non-printable character on initialization.

Also, the uart driver was explicitly clearing TX/RX interrupts in
its isr, which was unnecessary, as the act of reading/writing
did that implicitly.

These fixes allow the cc32xx uart to work with  [... cut]
commit 759b987f (2018-11-19): drivers: serial: Update cc32xx uart interrupt handling

The Zephyr console and shell interrupt processing assumes
a TX interrupt is evoked upon first enabling the TX
interrupt via uart_irq_tx_enable.

This was not the case with the cc32xx uart, coming out of
reset, with FIFO's disabled.

The only way found to achieve this behavior is to fill
the fifo with a non-printable character on initialization.

Also, the uart driver was explicitly clearing TX/RX interrupts in
its isr, which was unnecessary, as the act of reading/writing
did that implicitly.

These fixes allow the cc32xx uart to work with [... cut]
closed 2018-11-19 by galak