zephyrproject-rtos/zephyr #11213: NFFS: Handle unexpected Power Off

MitterdorferMathias, 2018-11-08:
Can NFFS handle unexpected power failures (either brownout reset - NRF52832, or loss of VCC)?

We encountered a corrupt nffs file system (see attachment).

It is not possible to unlink existing files (error code 3 - Invalid offset).

@nvlsianpu Can you please comment on this

[nffs_file_system_corrupt.bin.gz](https://github.com/zephyrproject-rtos/zephyr/files/2561937/nffs_file_system_corrupt.bin.gz)

nvlsianpu, 2018-11-09:
Apparently looks like not. As far i know there is any routine for checking integrity of file in nffs while the FS starts. This should be easy to add, also recovery previous files sate should be easy to do (as only need to restore previous version of the file inode).

Unfortunately what you encountered is not only nffs issue we have.
https://github.com/apache/mynewt-nffs/issues/10

MitterdorferMathias, 2018-11-09:
@nvlsianpu Thanks for the reply.

I think this would be a good feature to have.

nvlsianpu, 2020-01-27:
NFFS support was removed #21793.

closed 2020-01-27 by nvlsianpu