zephyrproject-rtos/zephyr #11232: nrf82: reel_board: USB unable to send fragmented packets through Control endpoint

finikorg, 2018-11-09:
When testing issue with wpanusb I noticed that after Host sends more then 64 bytes through Control endpoint communication stalls.
@mike-scott

carlescufi, 2018-11-09:
Assigned to @pawelzadrozniak so he can take a look.

finikorg, 2018-11-09:
[image]

pawelzadrozniak, 2018-11-20:
I cannot reproduce the problem. CPU hangs in radio IRQ handler after "iwpan phy $PHY set channel 0 26".

finikorg, 2018-11-20:
> I cannot reproduce the problem. CPU hangs in radio IRQ handler after "iwpan phy $PHY set channel 0 26".

Do you mean Linux hangs? BTW: have you disabled Bluetooth? Looks like reel_board always include BT in every project ;)

jfischer-phytec-iot, 2018-11-20:
>Do you mean Linux hangs? BTW: have you disabled Bluetooth? Looks like reel_board always include BT in every project ;)

:smile: well, like other NRF based boards:
boards/arm/reel_board/reel_board_defconfig:24:CONFIG_BT_CTLR=y

carlescufi, 2018-11-21:
@finikorg 
>that always enabling Bluetooth, is this a bug?

Well, we do the same for all nRF-based boards. That said this might not be ideal, but in any case we would need to revisit all the boards, so this is outside the scope of this issue.

carlescufi, 2018-11-21:
>Is this correct that BT does not work with 802.15.4 and the board crashes when both are enabled?

Yes, this is correct since both access the radio and are not coordinated.