zephyrproject-rtos/zephyr #11250: 'net help' shell command leads to a fatal fault on sam_e70

tbursztyka, 2018-11-09:
uart:~$ net help
net - Networking commands
Options:
  -h, --help  :Show command help.
Subcommands:
  allocs  :Print network memory allocations.
  app     :Print network application API usage information.
  arp     :Print information about IPv4 ARP cache.
  conn    :Print information about network connections.
  dns     :Show how DNS is configured.
  gptp    :Print information about gPTP support.
  http    :Print information about active HTTP connections.
  iface   :Print information about network interfaces.
  ipv6    :Print information about IPv6 specific information and configuration.
  mem     :Print information about network memory usage.
  nbr     :Print neighbor information.
  ping    :Ping a network host.
  route   :Show network route.
  rpl     :Show RPL mesh routing status.
  stacks  :Show network stacks information.
  stats   :Show network statistics.
  tcp     :Connect/send/close TCP connection.
  vlan    :S***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x204003c0
Faulting instruction address = 0x0
Fatal fault in thread 0x204003c0! Aborting.

jukkar, 2018-11-09:
What sample application you are using here?

I tried "net help" in echo-server sample and I see no errors.

aurel32, 2018-11-10:
bisected to:

```
a89690d10f175cac792c829ed042c0e6f65eb4e2 is the first bad commit
commit a89690d10f175cac792c829ed042c0e6f65eb4e2
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Nov 6 15:34:11 2018 +0200

    shell: Add APIs to set number of arguments
    
    This allows the shell core to perform precheck before calling the
    handler which then can assume the number of arguments is correct.
    
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>

:040000 040000 de27efd5ded816083a17ccfaecb5acf25bcfcc31 cd208c117cceb20788ced161eb37484ba0e13a3d M      doc
:040000 040000 916c3871afef6373299d89ff7d4090a71bae2d24 e5980db374dbffff2db2558f0fa749d977b15cb7 M      include
:040000 040000 b28506ccd04e6c55f23f73a7cf178bd7df073835 7da4f08ab9314920cb28809628b6044824aaf6a3 M      subsys
```

@Vudentz, could you please have a look?

qianfan-Zhao, 2018-11-11:
samples/subsys/shell/shell_module example has the same issue when the 'device' command executing.

The fault messasge and bisect track as same as this one.

```
uart:~$ device 
device - Device commands
Options:
  -h, --help  :Show command help.
Subcommands:
  levels  :List configured devices by levels
  list    :List***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x20000634
Faulting instruction address = 0x0
Fatal fault in thread 0x20000634! Aborting.
```

```
➜  zephyr git:(a89690d10f) git bisect bad 
a89690d10f175cac792c829ed042c0e6f65eb4e2 is the first bad commit
commit a89690d10f175cac792c829ed042c0e6f65eb4e2
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Nov 6 15:34:11 2018 +0200

    shell: Add APIs to set number of arguments
    
    This allows the shell core to perform precheck before calling the
    handler which then can assume the number of arguments is correct.
    
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>

:040000 040000 de27efd5ded816083a17ccfaecb5acf25bcfcc31 cd208c117cceb20788ced161eb37484ba0e13a3d M      doc
:040000 040000 916c3871afef6373299d89ff7d4090a71bae2d24 e5980db374dbffff2db2558f0fa749d977b15cb7 M      include
:040000 040000 b28506ccd04e6c55f23f73a7cf178bd7df073835 7da4f08ab9314920cb28809628b6044824aaf6a3 M      subsys
➜  zephyr git:(a89690d10f) 
```

qianfan-Zhao, 2018-11-11:
Besides can we provide a way to report bug on NULL dereference at runtime?

PR #11288 (2018-11-11): subsys: shell: Print help message only if command doesn't available
Fixes: #11250
commit e3506832 (2018-11-11): subsys: shell: Print help message only if command doesn't available

Fixes: #11250
closed 2018-11-11 by nashif