zephyrproject-rtos/zephyr #11261: EFR32FG1P (ARM Cortex-M4) MCU fails to wake up from sleep within _sys_soc_suspend()

mnkp, 2018-11-09:
I'm implementing power management features on SiLabs efr32_slwstk6061a board with EFR32FG1P ARM Cortex-M4 MCU. To put the device to sleep I'm using vendor provided em_emu library module.

After performing all the tedious configuration of Energy Management Unit the em_emu library puts the device to sleep via the WFI instruction. Unfortunately, despite correctly configured wake up event the device never wakes up.

The power management is performed within _sys_soc_suspend() function as described in [System Power Management](https://docs.zephyrproject.org/latest/subsystems/power_management.html#system-power-management) chapter of Zephyr's documentation. At this point, the kernel has disabled interrupts.

The culprit is the kernel method of disabling interrupts which in case of ARM Cortex-M4 architecture is done using BASEPRI register. Since the wake up event has an interrupt priority equal to or lower than the value in BASEPRI register the CPU never sees it. To disable interrupts and allow CPU to wake up the kernel should use the PRIMASK register.

From the ARM "Cortex™-M4 Devices Generic User Guide" chapter "2.5.2 Wakeup from sleep mode":

> #### Wakeup from WFI or sleep-on-exit
> Normally, the processor wakes up only when it detects an exception with sufficient priority to
cause exception entry. Some embedded systems might have to execute system restore tasks after
the processor wakes up, and before it executes an interrupt handler. To achieve this set the
PRIMASK bit to 1 and the FAULTMASK bit to 0. If an interrupt arrives that is enabled and has
a higher priority than current exception priority, the processor wakes up but does not execute the
interrupt handler until the processor sets PRIMASK to zero.

In case the interrupts are disabled using the PRIMASK method before putting the EFR32FG1P MCU to sleep the device wakes up correctly.

andyross, 2018-11-29:
That sounds wrong for sure.  What's possible reason would there be for leaving any interrupts at all masked when issuing a WFI instruction?

Note that k_cpu_idle() has a funny API -- it's called with an irq_lock taken, but has to release it before returning.  Is it doing it in the wrong order on ARM (i.e. WFI then irq_unlock() instead of the reverse)?

That said, I can't imagine this is that simple.  If the bug were that simple, WFI would never work for anything...

andyross, 2018-11-29:
Yeah, this is the relevant code from arch/arm/core/cpu_idle.S: it looks correct to me, affirmatively setting BASEPRI to zero before hitting the wfi.  How do you know that interrupt priority is the problem?  Are you hitting a WFI from some other place in the code?

```
#if defined(CONFIG_ARMV6_M_ARMV8_M_BASELINE)
        cpsie i
#elif defined(CONFIG_ARMV7_M_ARMV8_M_MAINLINE)
        /* clear BASEPRI so wfi is awakened by incoming interrupts */
        eors.n r0, r0
        msr BASEPRI, r0
#else
#error Unknown ARM architecture
#endif /* CONFIG_ARMV6_M_ARMV8_M_BASELINE */

        wfi
```

```

mnkp, 2018-11-30:
My issue concerns strictly _sys_soc_suspend() function, part of the System Power Management interface. In this case k_cpu_idle() is never called, rather the SoC provides its own implementation of an idle function.

Setting a breakpoint at the entrance of _sys_soc_suspend() function I can see that the BASEPRI register has value 0x20. In case of EFR32FG1P SoC NUM_IRQ_PRIORITY_BITS=3. So only three MSB bits are to be taken into account. The effective BASEPRI value is 1.

This is to be expected, as according to the _sys_soc_suspend() function documentation "When the kernel is about to go idle, the power management subsystem calls the _sys_soc_suspend() function [...]. At this point, the kernel has disabled interrupts and computed the maximum time the system can remain idle."

The only problem is that the kernel is disabling interrupts using an incorrect method. Rather than setting BASEPRI value to 1 it should set PRIMASK bit to 1.

andyross, 2018-11-30:
I guess I'm not understanding the complaint.  It's not the kernel's job to match interrupt masking to the needs of the HAL.  The kernel uses BASEPRI for interrupt locking, documents that, and calls the suspend function (not the idle loop, sorry, I misunderstood that part) in the state it claims, correctly.

If the platform has different requirements, it's surely the platform's job to implement that, no?

mnkp, 2018-12-01:
Obviously, I may have misunderstood the concept. However, for now, I'm quite confident that the implemented behavior of the kernel's code is not the intended one. To clarify that I would like to ask two questions:
1. What the kernel hopes to achieve by disabling interrupts using BASEPRI register?
2. Is user expected to enable interrupts before calling the WFI instruction?

andyross, 2018-12-01:
It's for mutual exclusion.  You absolutely have to have interrupt locking as an OS primitive.  This is the mechanism chosen (it's true that ARM has several -- I'm not expert enough to tell you why BASEPRI specifically was chosen as the irq_lock() implementation).

In particular, an idle/suspend implementation absolutely has to be atomic -- you can't decide to go to idle with interrupts enabled, because one might arrive (maybe with a bluetooth packet) too late (i.e. after the decision to idle has been made) and thus not get serviced correctly.

The job of the idle implementation (both k_cpu_idle() and _sys_soc_suspend()) is to atomically release the lock that is already held and enter the idle state.  That may involve some custom assembly, especially if the locking mechanisms expected by the HAL layer being invoked are different than the one the kernel is using.

mnkp, 2018-12-03:
> In particular, an idle/suspend implementation absolutely has to be atomic -- you can't decide to go to idle with interrupts enabled

Thanks for the detailed explanation. I'm afraid though it confirms that the current implementation of the kernel's code is incorrect. The only way to go to idle state with interrupts disabled is to follow the official ARM guidelines as detailed in a short excerpt from ARM "Cortex™-M4 Devices Generic User Guide" chapter "2.5.2 Wakeup from sleep mode" which I've included in the original description of this issue. To put it simply we have to enter the idle mode with PRIMASK bit set to 1.

Incidentally, this is what implementation of k_cpu_atomic_idle in zephyr/arch/arm/core/cpu_idle.S is doing. It sets PRIMASK bit to 1, BASEPRI to 0, calls WFE then restores the BASEPRI to an old value and sets PRIMASK bit to 0. We cannot however expect that user implementing _sys_soc_suspend() will know that they should code this specific sequence of events. It's undocumented, non-standard and fairly speaking quite bizarre.

To restate my complaint. When kernel calls _sys_soc_suspend() function the BASEPRI should be set to 0 and PRIMASK bit to 1.

mnkp, 2018-12-10:
Just to clarify possible misunderstanding. The behavior described in this issue is not vendor specific but rather common to all Cortex-M3 and higher (ARMv7-M) devices.

andyross, 2018-12-17:
To clarify, the requirement here is one of API conformance.  So ignore everything about BASEPRI and PRIMASK for a second. Here are the rules:

1. The Zephyr idle loop disables interrupts using irq_lock().
2. It then calls either k_cpu_idle() (a function provided by the architecture) or _sys_soc_suspend() (which comes from the SoC layer) to effect the actual idle.
3. On return from either function, **the IRQ lock is assumed to have been internally released**.

So... this is a weird API (among other things, the "key" returned by irq_lock(), which is required to be passed back to unlock in all other cases, never gets handed to the idle code which just assumes the relevant interrupt state is OK).  But it's our API nonetheless.

Add that to the following detail that is ARM-specific:

1. irq_lock() is implemented by setting BASEPRI to _EXC_IRQ_DEFAULT_PRIO, which is chosen to mask out all Zephyr-handled interrupts.

So that tells you that on entry to _sys_soc_suspend(), BASEPRI will be nonzero and PRIMASK has all interrupts unmasked (i.e. it's unused by zephyr).  On exit, BASEPRI must be zero and PRIMASK must be unchanged.

I understand that's not the interface that your HAL layer "wants". But it's the interface we have and that it must follow.  And there's no reason you can't add the ~4 instructions it would take to set and unset PRIMASK appropriately to your requirements.

mnkp, 2018-12-17:
> 3\. On return from either function, **the IRQ lock is assumed to have been internally released**.

I was wondering if this may be the case but then there was the obvious hindrance

> among other things, the "key" returned by irq_lock(), which is required to be passed back to unlock in all other cases, never gets handed to the idle code

Doesn't it imply that the current API design is broken?

> So that tells you that on entry to _sys_soc_suspend(), BASEPRI will be nonzero and PRIMASK has all interrupts unmasked (i.e. it's unused by zephyr). On exit, BASEPRI must be zero and PRIMASK must be unchanged.

We are forcing user to deal with low level, critical, implementation specific parts of the kernel code. This is really non obvious and not easy to infer.

> And there's no reason you can't add the ~4 instructions it would take to set and unset PRIMASK appropriately to your requirements.

This code is common to all implementations of `_sys_soc_suspend()` function for Cortex-M3 and higher devices. Even from the 'clean code' point of view it should be implemented by the calling function.

The design would be cleaner if either the kernel didn't lock the interrupts before calling `_sys_soc_suspend()` function or if we locked the interrupts in the proper manner. Locking the interrupts the wrong way on ARM devices just because the kernel code was written for other architectures where it doesn't cause any issues is not a good approach.

mnkp, 2018-12-17:
One way to fix the issue would be to have Zephyr idle loop disable interrupts using `irq_lock_before_sleep()` function (to be implemented). On architectures other than ARM it would map to `irq_lock()`.

The issue on how to call `irq_unlock()` without having a key argument remains.

andyross, 2018-12-17:
> Doesn't it imply that the current API design is broken?

Aesthetically, yes.  But in practice the lock is taken only in the one spot in the idle loop where we know the state.  Yes, it should be fixed.  It doesn't break anything in the meantime.

> We are forcing user to deal with low level, critical, implementation specific parts of the kernel code.

No... we're forcing the implementor of the SoC-specific suspend implementation to understand the details of the kernel implementation.  I don't see that this is so weird.  This isn't app code here :)

> This code is common to all implementations of _sys_soc_suspend() function for Cortex-M3 and higher devices

Sounds like a good opportunity for abstraction into an ARM SoC library routine then.

> The design would be cleaner if either the kernel didn't lock the interrupts before calling

That's not possible.  The decision to suspend needs to be atomic, otherwise we could decide the system was idle and begin calling _sys_soc_suspend() just as an interrupt occurred that (for example) woke up a thread to handle I/O.  We'd then return into the idle thread when had work to do!

> Locking the interrupts the wrong way on ARM devices just because the kernel code was written for other architectures

I still don't understand why you think BASEPRI is "the wrong way".  It's one of several alternatives, there's nothing preventing us from using PRIMASK, and the choice has nothing to do with other architectures.  If you really feel strongly that irq_lock() needs to work differently, you can try to take it up on zephyr-devel or with the ARM maintainers.

mnkp, 2018-12-17:
> I still don't understand why you think BASEPRI is "the wrong way".

The problem with the BASEPRI is that if we use it before calling WFI instruction the device will never wake up. We essentially mask out all the interrupt sources that we could use to wake the device up. The only way to go to idle state with interrupts disabled (and have a chance to wake up) is to use PRIMASK.

mnkp, 2018-12-17:
I understand that we should use BASEPRI in all other cases, no doubts there. It's just not appropriate when we want to put the device to sleep.

ramakrishnapallala, 2019-02-18:
The board seems similar to nRF boards which @pizi-nordic is working on.

ioannisg, 2019-10-24:
@mnkp @galak do you think we need to come back to this old  ARM issue?
Has this been addressed in any way in the last 12 months? If not, should we try to address it?

mnkp, 2019-10-26:
After the redesign of the Power Management subsystem the issue is less critical since the sensitive part of the power down / up code doesn't have to be written by the user any more. Instead this is handled by the Zephyr in-tree code located in the soc/ folder. 

Nevertheless all the issues described in this PR are still there:
- incorrect usage of BASEPRI when putting ARM devices to sleep.
- requirement that `sys_set_power_state` function will unlock interrupts using `irq_unlock` even if it is not passed the necessary `key` argument.

During the redesign `_sys_soc_suspend` function was removed, now `sys_set_power_state` is responsible for managing entry to/exit from a power state at the SoC level.

ioannisg, 2020-03-13:
@mnkp could you please clarify:
> incorrect usage of BASEPRI when putting ARM devices to sleep.

Does this concern arch_cpu_idle() for ARM, as well, and if so, why?

ioannisg, 2020-03-13:
> From the ARM "Cortex™-M4 Devices Generic User Guide" chapter "2.5.2 Wakeup from sleep mode":
>>Wakeup from WFI or sleep-on-exit.......

BTW, the posted comment from the user-guide does not mean that this is the only way for the WFI to work. That is, setting PRIMASK to 1 before WFI is not mandatory. This is only to be able to execude code _after_ WFI but _before_ the ISR.

stephanosio, 2020-03-17:
> The problem with the BASEPRI is that if we use it before calling WFI instruction the device will never wake up.

@mnkp Could you try adding `isb` before `wfi` and see if that makes a difference? (#23511)

ioannisg, 2020-04-02:
>- incorrect usage of BASEPRI when putting ARM devices to sleep.
>- requirement that sys_set_power_state function will unlock interrupts using irq_unlock even if it is not passed the necessary key argument.

I believe the first bullet is now addressed, at least for k_cpu_idle().
As for the second bullet, I don't see this requirement in the function documentation.

Closing this for now; @mnkp pls, re-open if needed.

closed 2020-04-02 by ioannisg

zjli-2019, 2024-06-15:
Below is an incomplete summary of efforts to fix this. It is far better to handle BASEPRI/PRIMASK properly and things can be kept DRY.

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/nxp/rw/power.c#L37

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/silabs/common/soc_power_pmgr.c#L49

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/microchip/mec/mec172x/power.c#L49

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/ti/simplelink/cc13x2_cc26x2/power.c#L69

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/nxp/imxrt/imxrt10xx/power.c#L90

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/nxp/imxrt/imxrt6xx/cm33/power.c#L29

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/ambiq/apollo4x/power.c#L24

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/renesas/smartbond/da1469x/power.c#L22

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/nxp/imxrt/imxrt5xx/cm33/power.c#L27

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/nuvoton/npcx/common/power.c#L146

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/ti/simplelink/cc13x2x7_cc26x2x7/power.c#L70

https://github.com/zephyrproject-rtos/zephyr/blob/3b85e8b8ea27c11d533c188cccf6e795907f10ed/soc/nxp/imxrt/imxrt11xx/power.c#L246