zephyrproject-rtos/zephyr #11276: nrf51_pca10028:tests/kernel/interrupt/arch.interrupt registers IRQ 24 twice

andrewboie, 2018-11-09:
As noted in build failures for #11273 which checks for this.
IRQ_CONNECT() shouldn't ever be called more than once on the same IRQ line, it's undefined which one will actually get registered.

```
[134/139] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       53668 B       256 KB     20.47%
            SRAM:       11372 B        32 KB     34.70%
        IDT_LIST:         152 B         2 KB      7.42%
[135/139] Generating linker_pass_final.cmd
[136/139] Generating isr_tables.c
FAILED: cd /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/out-2nd-pass/nrf51_pca10028/tests/kernel/interrupt/arch.interrupt/zephyr && /opt/sdk/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-objcopy -I elf32-littlearm -O binary --only-section=.intList /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/out-2nd-pass/nrf51_pca10028/tests/kernel/interrupt/arch.interrupt/zephyr/zephyr_prebuilt.elf isrList.bin && /usr/bin/python3 /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py --output-source isr_tables.c --kernel /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/out-2nd-pass/nrf51_pca10028/tests/kernel/interrupt/arch.interrupt/zephyr/zephyr_prebuilt.elf --intlist isrList.bin   --sw-isr-table --vector-table
gen_isr_tables.py: multiple registrations for irq 24 (0x18)
Traceback (most recent call last):
  File "/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py", line 295, in <module>
    main()
  File "/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py", line 287, in main
    error("multiple registrations for irq %d (0x%x)" % (irq, irq))
  File "/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/common/gen_isr_tables.py", line 34, in error
    raise Exception()
Exception
ninja: build stopped: subcommand failed.
```

andrewboie, 2018-11-09:
Assigning to @ajaykish as he wrote the test, which is registering testcase interrupts which appear to clash with real IRQs on this board.

carlescufi, 2018-11-20:
@ajaykish are you going to take a look at this, or should @anangl?

ajaykish, 2018-11-23:
@carlescufi  you can ask @anangl, to have a look at this. I am quite busy with some high priority task. Will work on this once done with current task.

ioannisg, 2018-12-14:
@carlescufi Will do

nashif, 2019-01-30:
@ioannisg ping

ioannisg, 2019-02-04:
@andrewboie I can not reproduce the error using the diff in gen_isr_tables from #11273.

ioannisg, 2019-02-04:
So, it looks to me that in nRF51 boards, where BT is enabled by default, SWI5_IRQn is registered with IRQ_CONNECT, which corresponds to line 25, not line 24.
This might have changed, recently, can you confirm @cvinayak ?

ioannisg, 2019-02-04:
[Continuing]
As far as I could see, default builds for nRF51-based boards enable Bluetooth controller and use SWI5 (line 25) for BT. This is because the LL split is not enabled by default (@cvinayak pls, confirm).

Therefore, line 24 is now available for the test, so, this issue can be closed as "invalid".

ioannisg, 2019-02-04:
Closing this - not an issue anymore (unless and until the default IRQ registration changes again for nrf51)

closed 2019-02-04 by ioannisg