zephyrproject-rtos/zephyr #11296: Possible ways to implement clock synchronisation over BLE

m-nilsson, 2018-11-12:
We need to implement precision clock synchronisation between two Nordic NRF52 devices. I have seen the example using the Timeslot API with the Nordic SDK. Is there some similar technique that can be used with Zephyr? I am thinking that there might be even more possibilities to achieve something like that with the Zephyr infrastructure since there are no closed source involved.

For example, if there is a way to retrieve a timestamp or counter value at the point where a BLE packet is actually sent or received, then we could develop synchronisation a bit like the way PTP works on ethernet networks. Preferably there could be a callback that also had access to the data payload of the message sent or received.

Or are there other possibilities?

If there is no suitable mechanism in the current API, maybe you could suggest an appropriate place in the code to put these things. Thanks!

carlescufi, 2018-12-07:
@cvinayak could you please take a look at this question?

m-nilsson, 2018-12-18:
@cvinayak Thanks a lot!

carlescufi, 2018-12-18:
Solved

closed 2018-12-18 by carlescufi

lanwer-en, 2020-01-09:
> @cvinayak Thanks a lot!

I am facing exactly the same problem as I want to synchronize 2 nRF52 devices without using the timeslot API from the Nordic SDK and use Zephyr instead. It would be really helpful if you could share your code.