zephyrproject-rtos/zephyr #11301: ieee802154

ekarlso, 2018-11-12:
KW41Z with Echo Server|Client hard faults

**To Reproduce**
Just flash a kw41z with echo-server or client @ master zephyr and look at console
cmake -GNinja  -DBOARD=frdm_kw41z -DCONF_FILE="boards/frdm_kw41z.conf prj.conf" ../
cp zephyr/zephyr.bin /media/<user>/FRDM-KW41ZJ/

**Expected behavior**
Should work out of the box.

**Impact**
Showstopper to test 802154 between devices.

**Toolschain**
SDK 0.9.5 and 7.3.1
master

ekarlso, 2018-11-12:
cc @MaureenHelm

pfalcon, 2018-11-13:
cc @dleach02 

@ekarlso, it always makes sense to include the exact HEAD revision of the master you used.

pfalcon, 2018-11-13:
> Just flash a kw41z with echo-server or client @ master zephyr and look at console

And people will start with looking at this bug, and it doesn't provide enough info on what happens...

tbursztyka, 2018-11-13:
@pfalcon because this not more info that what he says in the beginning "hard faults". Directly. I tried and there is nothing but the hard fault.

ekarlso, 2018-11-13:
@pfalcon basically what @tbursztyka says, it goes nuclear and faults (Nothing really useful in the console)

ekarlso, 2018-11-13:
@mnkp @chrta Do we have a overview of what MCU's support DCDC?

tbursztyka, 2018-11-14:
Thanks @MaureenHelm for looking into this. This is indeed a surprising issue since all attributes in this struct are all pointers thus 4 bytes each so I would have assumed linker would not have much to do here for alignment.

SebastianBoe, 2018-11-14:
> I'm not sure this is the right fix though. @SebastianBoe, could some of the recent build system and linker changes have caused this?

I'm not aware of any changes that can affect this. A git bisect from someone with a KW417 (or any M0 board with enough RAM) would be helpful.

mnkp, 2018-11-14:
I may have found the root cause. As we know any variable or constant of type `struct ieee802154_radio_api` may be placed at an unaligned memory address (even if all the members are aligned at 4 bytes).

When we access the API via the predefined inline functions
```
static inline int ieee802154_set_tx_power(struct net_if *iface, s16_t dbm)
{
        const struct ieee802154_radio_api *radio =
                net_if_get_device(iface)->driver_api;

        return radio->set_txpower(net_if_get_device(iface), dbm);
}
```
the access will be slow but should work. However that does not hold true for the first field `struct net_if_api iface_api;`. As the comment suggests
```
	 * Mandatory to get in first position.
	 * A network device should indeed provide a pointer on such
	 * net_if_api structure. So we make current structure pointer
	 * that can be casted to a net_if_api structure pointer.
	struct net_if_api iface_api;
```
the iface_api is accessed directly by the networking code and ultimately the __packed is 'casted away'. I didn't track down the exact place where it happens though.

As far as I can tell the __packed for `struct ieee802154_radio_api` is not required. We could simply remove it.

dleach02, 2018-11-14:
while I agree that the structure definition doesn't need the __packed directive I don't quite understand how its presence affects the alignment of instances of this structure? Is this because the __packed directive causes GCC to believe the natural alignment of the structure is 1 byte aligned? 

The __packed directive has been on this structure for a long time so why is this just now showing up?

pfalcon, 2018-11-14:
> The __packed directive has been on this structure for a long time so why is this just now showing up?

Source code stays, compilers come and go, acquiring crazier and crazier features...

tbursztyka, 2018-11-15:
__packed is indeed useless here (probably a c/p left over when I started this api... hum). That said, the new behaviour is indeed suspicious, well I am going to send a patch removing this. Thanks @MaureenHelm and @mnkp for looking at this.

closed 2018-11-15 by nashif

dleach02, 2018-11-16:
My fear is that we don't really know what is causing this alignment problem and taking __packed off doesn't really explain why it fixes the problem. The result is that this problem may crop up someplace else..

pfalcon, 2018-11-16:
I for one 100% sure that it's caused by __packed and 100% sure that it was the right fix (in general, there should be no __packed on any structure which contains a pointer).

dleach02, 2018-11-19:
I understand. Just not sure why this just now popped up. Maybe this is a new compiler vs old compiler.

My understanding of packed structure is to pack the contents of the structure but this setting seemed to carry to the alignment of the whole structure which seems problematic. I did see a long running bug in the gcc bug database about packed and bus faults with unaligned access but it wasn't directly because the base structure was put on an unaligned address.

mnkp, 2018-11-19:
We didn't remove __packed because we noticed that it solved the issue but because it shouldn't be there in the first place. There is no use for this __packed, also as @pfalcon mentioned 'in general, there should be no __packed on any structure which contains a pointer'. The redundant attribute was overlooked during the code review.

With __packed not only structure fields but also the structure itself can be unaligned. Looking at the map file of 'echo_server' sample application compiled for frdm_kw41z board we have
```
 .data.kw41z_radio_api
                0x000000002000ad59       0x28 zephyr/libzephyr.a(ieee802154_kw41z.c.obj)
```
so `kw41z_radio_api` variable holding `struct ieee802154_radio_api` is at unaligned address.

If the initial check-in happened to place `kw41z_radio_api` variable after some other which size was multiple of 4 it would take a very specific, random modification of code to have linker shuffle variable locations and place `kw41z_radio_api` at an unaligned address. This bug could have gone unnoticed for a long time despite multiple code changes.

The observed hard fault might have been caused by the gcc bug as @dleach02 suggested but most likely it happened when networking stack was trying to access iface_api field. Most of the accesses to the `struct ieee802154_radio_api` are done in a safe way via API wrapper functions with the exception of `struct net_if_api iface_api` field which is accessed directly by the networking stack in a way that assumes an aligned address. E.g. here:
```
subsys/net/ip/net_if.c:	const struct net_if_api *api = net_if_get_device(iface)->driver_api;
```
Only a person having the board could verify it for sure though, I reckon, there is no need to dig into this issue any further.

dleach02, 2018-11-19:
Fair enough. I've just never seen the packed modifier affect the alignment of the variable. Just the internal fields of the structure.

PR #11398 (2018-11-15): net/ieee802154: Remove unnecessary __packed keyword on API struct
It is useless first, and for some reason now generate a bug on kw41z.

Fixes #11301
commit 2a0669e5 (2018-11-15): net/ieee802154: Remove unnecessary __packed keyword on API struct

It is useless first, and for some reason now generate a bug on kw41z.

Fixes #11301