zephyrproject-rtos/zephyr #11310: Wifi scan crashes on disco_l475_iot1

erwango, 2018-11-12:
**Describe the bug**

When running samples/net/wifi with disco_l475_iot1, crash happens.

**To Reproduce**
Steps to reproduce the behavior:
0. cd samples/net/wifi
1. mkdir build; cd build
2. cmake -DBOARD=disco_l475_iot1
3. make
4. See error

**Expected behavior**
Scan is performed and returns list of available APs

**Impact**
Prevent use of Wifi on board

**Screenshots or console output**
```
[00:00:13.154,832] <err> wifi_eswifi_bus_spi.eswifi_spi_request: CMD ready timet

[00:00:13.154,832] <err> wifi_eswifi_bus_spi.eswifi_spi_request: CMD ready timet

[00:00:13.154,835] <err> wifi_eswifi_core.eswifi_iface_init: Unable to read MACs
[00:00:13.154,835] <err> wifi_eswifi_core.eswifi_iface_init: Unable to read MACs
[00:00:13.154,838] <err> net_if.init_iface: {assert: 'api->send' failed}        
[00:00:13.154,838] <err> net_if.init_iface: {assert: 'api->send' failed}        
***** Reserved Exception ( -16) *****                                           
***** Hardware exception *****                                                  
Current thread ID = 0x20001d8c                                                  
Faulting instruction address = 0xaaaaaaaa                                       
Fatal fault in essential thread! Spinning... 
```

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain Zephyr SDK
 - SHA1: b5ceade03c28448357736846bd7d043787961ffd

**Additional context**
Add any other context about the problem here.

erwango, 2018-11-12:
@loicpoulain, was working recently but crashing now, can you have a look?
Thanks

loicpoulain, 2018-11-15:
@erwango, just checked it, I thought #10372 was merged, but actually not. It's requested to prevent spi error.

tbursztyka, 2018-11-15:
reopened #10372, sorry

erwango, 2018-11-19:
Tested. Working, thanks!
Closing the ticket

closed 2018-11-19 by erwango