zephyrproject-rtos/zephyr #11333: sockets: Implement select() call

pfalcon, 2018-11-13:
Now that sockets are backed by file descriptors, i.e. we know they're small integers, we can implement select().

pfalcon, 2018-11-13:
@GAnthony, @rlubos : FYI

pfalcon, 2018-11-13:
Actually, all this time everyone (?, perhaps just me) assumed that select, well, FD_SET, would need to be implemented in a classic (?) way using bitsets. Ahem, it doesn't have to! We could implement it long ago, with FD_SET backed by struct pollfd[]. Of course, then FD_ISSET() would need to loop over this array, which is O(n), but given that the array would be small anyway, nobody would care. And those who care about performance, shouldn't use select() in the first place.

So, I guess we should implement it like that now.

pfalcon, 2018-12-03:
@GAnthony, @rlubos: So, I assume there're no criticism/concerns. I otherwise would plan this implementation yet for this year.

@jukkar, @tbursztyka: Also if you're interested, I try not to ping you on socket details, maybe I should.

If there's something to discuss, there's opportunity to do that at tomorrow's networking forum.

GAnthony, 2018-12-04:
@pflacon, Thanks, select() would be good to have.

PR #12049 (2019-01-24): net: sockets: Implement select() call.
It's implemented on top of poll() anyway, and the current
implementation of fd_set uses array of fd's underlyingly, which
leads to O(n) complexity for FD_SET() and friends.

The purpose of select() implementation is to allow to perform
proof-of-concept port of 3rd-party code to Zephyr quickly. For
efficiency, poll() should be used instead.

Fixes: #11333
commit 3a0a9944 (2019-01-24): net: sockets: Implement select() call.

It's implemented on top of poll() anyway, and the current
implementation of fd_set uses array of fd's underlyingly, which
leads to O(n) complexity for FD_SET() and friends.

The purpose of select() implementation is to allow to perform
proof-of-concept port of 3rd-party code to Zephyr quickly. For
efficiency, poll() should be used instead.

Fixes: #11333
closed 2019-01-24 by jukkar