zephyrproject-rtos/zephyr #11343: Clarification of LOG_MODULE_REGISTER placement

derchrismakaio, 2018-11-13:
I'm trying to implement a driver which needs to include net_if.h (because I want to use **struct net_if_api**)
net_if.h includes net_core.h which has (ifdef NET_LOG_LEVEL) _LOG_MODULE_REGISTER(LOG_MODULE_NAME);_

When I now try to include my own header in my implementation .c file, I can't do a _LOG_MODULE_REGISTER(app, 4);_
as it will give redefinition warnings.

Here's the compiler output:
`error: redefinition of '__log_current_const_data'`
`error: redefinition of '__log_current_dynamic_data'`
`error: redefinition of '__log_level'`

This is basically my compilation code for testing:

**my_driver.c**

```
#include "../core/drivers/my_driver.h"
#include <logging/sys_log.h>
LOG_MODULE_REGISTER(my_driver, 4);
```

**my_driver.h**

```
#include <kernel.h>
#include <device.h>
#include <net/net_if.h>

struct my_driver_cb {
    void (*my_driver_ready)(bool success);
    void (*my_driver_attached)(bool success);
};

typedef int (*my_driver_register_cb)(struct my_driver_cb* cb);
typedef int (*my_driver_connect)(void);

struct modem_driver_api {
	struct net_if_api iface_api;

    my_driver_register_cb register_cb;
    my_driver_connect connect;
};
```

jarz-nordic, 2018-11-13:
I see sys_log header. It looks like legacy code. @nordic-krch may you please take a look?

derchrismakaio, 2018-11-13:
ah sorry, that was a copy & paste mistake, I'll edit the sources

derchrismakaio, 2018-11-13:
so, it looks like net_core.h depends on #define LOG_MODULE_NAME

If I change 
```
#include "../core/drivers/my_driver.h"
#include <logging/log.h>
LOG_MODULE_REGISTER(my_driver, 4);
```

to

```
#define LOG_MODULE_NAME my_driver
#include <logging/log.h>
```
(note: removed LOG_MODULE_REGISTER as it is in done in inherently included net_core.h)

it's working fine, but all net-specific logging will inherit my LOG_MODULE_NAME "my_driver", I guess

mike-scott, 2018-11-13:
Apparently there is also confusion on the order of #define LOG_LEVEL vs. #include <logging/log.h>: https://github.com/zephyrproject-rtos/zephyr/pull/11352

nordic-krch, 2018-11-14:
This is rather an issue in net subsystem: usage of `LOG_MODULE_REGISTER` in header which is frequently included by API headers. There are a log of workarounds for that in the code (example https://github.com/zephyrproject-rtos/zephyr/blob/9d14874db18e0d5b87bb4af748e96bda474a4417/subsys/usb/class/netusb/function_rndis.c#L15). @jukkar do you plan to fix that?

derchrismakaio, 2018-11-14:
I'm not 100% sure but it looks like this is really a critical bug because since I made my changes I know get BUS FAULTS on some occurences of LOG_DBG

jukkar, 2018-11-28:
Not sure what is the issue here, but net_core.h will only call `LOG_MODULE_REGISTER(LOG_MODULE_NAME);` if NET_LOG_LEVEL is defined. The NET_LOG_LEVEL is only enabled in various network related .c files. There should be no issue including net_if.h anywhere in the zephyr code base.

nordic-krch, 2018-11-28:
@jukkar note that there is:
```
#if !defined(NET_LOG_LEVEL)
#define NET_LOG_LEVEL CONFIG_NET_DEFAULT_LOG_LEVEL
#endif /* !NET_LOG_LEVEL */
```
at the top of `net_core.h`. 

There are workarounds for that in the code already (see example i've pasted above).

I think @rlubos also hit that issue recently.

I would recommend to move `LOG_MODULE_REGISTER` to .c files, especially that now compact version is supported:
`LOG_MODULE_REGISTER(module_name, CONFIG_module_LOG_LEVEL)`

jukkar, 2018-11-29:
> I would recommend to move `LOG_MODULE_REGISTER` to .c files, especially that now compact version is supported:
> `LOG_MODULE_REGISTER(module_name, CONFIG_module_LOG_LEVEL)`

Ok, good to know. I will try to figure out some sense to this and will send a PR asap.

jukkar, 2018-11-29:
@nordic-krch I have done this in various .c files
```
#include <logging/log.h>
LOG_MODULE_REGISTER(net_xxx, CONFIG_xxx_LOG_LEVEL);
```

and removed the log level setting from net_core.h. But now I see
```
zephyr/include/logging/log_core.h:228:13: error: ‘__log_level’ undeclared (first use in this function); did you mean ‘__locale_t’?
  (_level <= __log_level) &&         \
             ^~~~~~~~~~~
```

Am I suppose still declare LOG_LEVEL manually, or should the LOG_MODULE_REGISTER() do that automatically?

commit 86689030 (2018-12-07): net: Clarify logging in networking code

Remove network specific default and max log level setting
and start to use the zephyr logging values for those.

Remove LOG_MODULE_REGISTER() from net_core.h and place the
calls into .c files. This is done in order to avoid weird
compiler errors in some cases and to make the code look similar
as other subsystems.

Fixes #11343
Fixes #11659
closed 2018-12-07 by jukkar