zephyrproject-rtos/zephyr #11368: Convert pwm_nrf5_sw to device tree

galak, 2018-11-14:
Convert the pwm_nrf5_sw driver to utilize device tree.

galak, 2018-11-14:
By converting this driver we can remove explicit inclusion of <board.h> in samples/boards/microbit/sound & samples/boards/microbit/pong, and samples/bluetooth/mesh_demo

carlescufi, 2018-12-11:
@gamnes since you've done `pwm_nrfx`, could you complete the transition by also moving `pwm_nrf5_sw` to DT?

gamnes, 2018-12-13:
Just some notes;
The include we get from board.h in sound/src/main.c, is the EXT_P0_GPIO_PIN - which is then used to define BUZZER_PIN, which is passed into the PWM driver.

If we are going to define the PWM SW outputs in DT; like we now do for the HW PWM in a SAMPLE_DT.overlay; then we should be able to get rid of the <board.h> include.

There isn't anything stopping us from changing SW PWM GPIOs at run-time, but if we want this SW implementation to match the current nRF HW PWM implementation, then I guess it would make sense to define the SW PWM GPIOs in DT as well.

Because the buzzer is an external component to the MICROBIT board, I assume this buzzer DT definition will be placed in the SAMPLE_DT.overlay file (along with the SW PWM GPIO definition).

galak, 2018-12-13:
These would just go in the bbc microbit dts file.

gamnes, 2018-12-13:
But the BUZZER is an external component to the bbc microbit board; So if this is defined within the microbit DTS, then we are basically stating that every microbit board application (/sample) will use the SW PWM driver with an output on P0 for a BUZZER. Or have I misunderstood something there?

I guess it would make sense that every configuration of the BBC_MICROBIT board defines their own board DTS file - so if you are using the BBC_MICROBIT board with a BUZZER attached to EXT_P0, then you should make a new board.dts that defines SW PWM to output on P0. Is that the idea?

gamnes, 2019-01-09:
PR submitted, https://github.com/zephyrproject-rtos/zephyr/pull/12390, #12390. If there are any objections to how that is solved, we can discuss it there.

gamnes, 2019-02-12:
@galak : The PR i referred to above has been updated to revert some of the changes i made initially of moving SW PWM OUTPUT selection to DT. 
Now, as before, the SW PWM OUTPUTs are chosen at run-time, and not in DT - hence the board.h include in the samples you have referred to are kept as is to get hold of the EXT_PIN define. If you feel that is incorrect, please elaborate and describe how you would like to solve it.

closed 2019-03-12 by gamnes