zephyrproject-rtos/zephyr #11373: Move STM32 RTC to new counter API

erwango, 2018-11-14:
**Is your feature request related to a problem? Please describe.**
Move STM32 RTC driver to new counter API (cf #8340)

**Describe the solution you'd like**
Could look at https://github.com/zephyrproject-rtos/zephyr/pull/11347 as exemple of similar work

galak, 2018-11-14:
@ydamigos Is this something you might have some time to look into doing?

ydamigos, 2018-11-14:
@galak My time is limited but I'll give it a try,

erwango, 2018-11-28:
@ydamigos , have you started? I should find some time to check it this week.

ydamigos, 2018-11-28:
@erwango yes, I did some work on it. You could find it [here](https://github.com/ydamigos/zephyr/tree/stm32-rtc). It compiles but it doesn't work on target.

erwango, 2018-11-28:
ok, I finally started as well :-). It is [here](https://github.com/erwango/zephyr/tree/dev_rtc_counter_stm32). Not linking with Cube for some issue for now.
I took some options that could be discussed since new API doesn't fit nicely STM32 RTC IP, but out of our 2 initiatives we should manage to get through.

commit 5269136f (2019-02-07): drivers/counter: stm32: Provide STM32 RTC driver using counter API

Provides a STM32 RTC driver using new counter API.
Driver does not support wrap related functions (set, get, ..)

Fixes #11373
closed 2019-02-07 by nashif