zephyrproject-rtos/zephyr #11375: Enhance DTS yaml support to handle device being on different bus types

galak, 2018-11-14:
Currently we have sensors that can be connected over either I2C or SPI.  To handle this we have different compatible names like ("st,lsm6dsl-spi" & "st,lsm6dsl").  We really should only have a single compatible for the device ("st,lsm6dsl").

erwango, 2018-11-19:
I'm surprised by this request.
Node for spi bus slave or i2c bus slave will have different properties, how can we make map this to single compatible ? (which  mean a single driver will have to be able to handle both)

galak, 2018-11-19:
> I'm surprised by this request.
> Node for spi bus slave or i2c bus slave will have different properties, how can we make map this to single compatible ? (which mean a single driver will have to be able to handle both)

All of the properties that aren't bus specific should be identical for the same sensor like st,lsm6dsl.  I assume that if a sensor supports connecting over I2C or SPI that everything else is the same.

erwango, 2018-11-19:
Ok, so then, they can share a common "phy", but I don't how we can have common bus slaves.
So 2 bus slaves variants that share the same sensor phy node?

commit 2442382f (2018-12-07): scripts: extract_dts_includes: Support same sensor on different bus

There are a number of cases in which we a sensor can be either connected
on I2C or SPI.  We've been treating these cases as different compatiable
properties, but they really should use the same compatiable and just
determine the info based on the parent bus in the device tree.  We can
now support having two different binding files for the same compatiable
to handle the case of a sensor supporting either I2C or SPI as how its
connected.

We put "sensor" nodes in a bus specific dict bus_bindings[bus] that
we can than lookup lat [... cut]
closed 2018-12-07 by galak