zephyrproject-rtos/zephyr #11383: tests/drivers/i2s/i2s_api failing on sam_e70_xplained

andrewboie, 2018-11-14:
**Describe the bug**
This test defines two suites, i2s_loopback_test and i2s_states_test. i2s_loopback_test passes, but i2s_states_test is failing:
```
Running test suite i2s_states_test
===================================================================
starting test - test_i2s_tx_transfer_configure_1
PASS - test_i2s_tx_transfer_configure_1
===================================================================
starting test - test_i2s_rx_transfer_configure_1
PASS - test_i2s_rx_transfer_configure_1
===================================================================
starting test - test_i2s_state_not_ready_neg
[00:00:04.250,057] <dbg> dev_i2s_sam_ssc.i2s_sam_write: invalid state
[00:00:04.491,271] <dbg> dev_i2s_sam_ssc.bit_clock_set: freq = 256000
[00:00:04.505,136] <dbg> dev_i2s_sam_ssc.bit_clock_set: freq = 256000
[00:00:04.518,656] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: START trigger: invalid state
[00:00:04.518,664] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: DRAIN trigger: invalid state
[00:00:04.518,671] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: STOP trigger: invalid state
[00:00:04.518,678] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: DROP trigger: invalid state
[00:00:04.518,685] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: PREPARE trigger: invalid state
[00:00:04.518,690] <dbg> dev_i2s_sam_ssc.i2s_sam_read: invalid state
[00:00:04.518,702] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: START trigger: invalid state
[00:00:04.518,716] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: DRAIN trigger: invalid state
[00:00:04.518,724] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: STOP trigger: invalid state
[00:00:04.518,731] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: DROP trigger: invalid state
[00:00:04.518,736] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: PREPARE trigger: invalid state
[00:00:04.518,752] <dbg> dev_i2s_sam_ssc.i2s_sam_write: invalid state
PASS - test_i2s_state_not_ready_neg
===================================================================
starting test - test_i2s_state_ready_neg
PASS - test_i2s_state_ready_neg
===================================================================
starting test - test_i2s_state_running_neg
[00:00:04.638,157] <dbg> dev_i2s_sam_ssc.bit_clock_set: freq = 256000
[00:00:04.638,165] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: DRAIN trigger: invalid state
[00:00:04.638,174] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: STOP trigger: invalid state
[00:00:04.638,181] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: PREPARE trigger: invalid state
[00:00:04.638,189] <dbg> dev_i2s_sam_ssc.bit_clock_set: freq = 256000
[00:00:04.638,200] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: DRAIN trigger: invalid state
[00:00:04.638,206] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: STOP trigger: invalid state
[00:00:04.638,213] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: PREPARE trigger: invalid state
[00:00:04.655,082] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: START trigger: invalid state
[00:00:04.655,090] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: PREPARE trigger: invalid state
[00:00:04.655,102] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: START trigger: invalid state
[00:00:04.655,111] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: PREPARE trigger: invalid state

    Assertion failed at /home/apboie/projects/zephyr2/tests/drivers/i2s/i2s_api/src/test_i2s_states.c:33O

FAIL - test_i2s_state_running_neg
===================================================================
starting test - test_i2s_state_stopping_neg

    Assertion failed at /home/apboie/projects/zephyr2/tests/drivers/i2s/i2s_api/src/test_i2s_states.c:37)
RX START trigger failed
FAIL - test_i2s_state_stopping_neg
===================================================================
starting test - test_i2s_state_error_neg

    Assertion failed at /home/apboie/projects/zephyr2/tests/drivers/i2s/i2s_api/src/test_i2s_states.c:43)
RX START trigger failed
FAIL - test_i2s_state_error_neg
===================================================================
===================================================================
PROJECT EXECUTION FAILED
[00:00:04.743,897] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: START trigger: invalid state
[00:00:04.770,784] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: START trigger: invalid state
[00:00:04.799,466] <dbg> dev_i2s_sam_ssc.i2s_sam_trigger: START trigger: invalid state
```
This was undetected by CI due to an issue where if a test defines multiple suites, and the first suite to run passes, CI ignores the result of any subsequent suites.

**To Reproduce**
Build/flash the test on sam_e70_xplained, run on a board and observe console output

**Expected behavior**
All tests pass.

**Impact**
It means the driver has bugs.

aurel32, 2018-11-14:
Bisection leads to that commit:
```
a66967b0f2af45460062a6d26cba10622a20544e is the first bad commit
commit a66967b0f2af45460062a6d26cba10622a20544e
Author: Anas Nashif <anas.nashif@intel.com>
Date:   Mon Oct 8 18:03:38 2018 -0400

    driver: i2s: move to new logger
    
    Move drivers to new logger and change samples enabling logging in
    prj.conf.
    
    Signed-off-by: Anas Nashif <anas.nashif@intel.com>

:040000 040000 4437eaf973f1c1c9f013d95af25920fea4b39e3d 86ab5e1f99f2d3319343a7198241ba18931690e6 M      drivers
:040000 040000 338c22043c65c7eda42c7f8c95e123699051e0b0 1003ff4a574d2aaba3987cd6fa83919257a575fa M      tests
```

It looks like a bad interaction between the I2S driver and the logger. Reducing the log level to `inf` from `dbg` also makes the problem disappear.

andrewboie, 2018-11-14:
in my testing, it seems that this test case is VERY timing sensitive, adding printks in various places in the test, or stepping through test functions in a debugger, can cause them to fail

aurel32, 2018-11-14:
At least for the `test_i2s_state_running_neg` case, the test is submitting regularly blocks to the I2S driver, and thus expect it to stay it run state. This is done from the main thread, with priority 0. Unfortunately the log thread has priority -2 and takes some time to complete, causing a buffer underrun.

A workaround is to reduce the priority of the log thread to 0.

andrewboie, 2018-11-15:
Why is the log thread such high priority? Shouldn't it be set to the lowest preemptible priority?

aurel32, 2018-11-15:
The problem is that with the default UART backend, the log is dumping the data to the serial port in polling mode. Therefore when the log thread kicks in, it doesn't yield until the log messages are fully dumped, which might take a few hundred of milliseconds, causing the I2S underrun.

When using the shell backend instead, with `CONFIG_UART_INTERRUPT_DRIVEN=y`, the problem disappears, as the log thread yields between the UART interrupts.