zephyrproject-rtos/zephyr #11393: printk (RTT backend) from isr can deadlock the system

pdunaj, 2018-11-15:
**Describe the bug**
Calling printk from interrupt callback can deadlock the system if RTT backend is used.

The problem is caused by `SEGGER_RTT_LOCK` called from `rtt_console_out` (in `./drivers/console/rtt_console.c`). Console out is called directly from printk (see `_vprintk`) and can cause deadlock if called from an interrupt context.

**To Reproduce**
Steps to reproduce the behavior:
1. Build with printk enabled, `CONFIG_LOG_PRINTK` disabled, enable RTT backend.
2. At the same time log messages from thread and use printk from interrupt.
3. Observe system deadlock

**Expected behavior**
Printk should not deadlock the system under any circumstances.

**Impact**
Cannot use debug messages from interrupts with RTT backend.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
rev:`f2d8d93269ddce9dd7629c8a8c6d38d511c3d399`

**Additional context**
N/A

pdunaj, 2018-11-15:
btw `CONFIG_LOG_PRINTK` should be usable workaround

carlescufi, 2018-11-20:
CC @pkral78

pkral78, 2018-11-20:
Well, I don't see `CONFIG_LOG_PRINTK` as a workaround rather than proper solution, what is purpose to use rtt console (seems deprecated to me) instead of logger backend when RTT logger backend is enabled at same time ?

But anyway I agree that printk must be usable from isr context, so if use of rtt console is really a must, there is proposed solution:
1.) remove locking macros from `rtt_console.c` completely as they are not usable from isr context
2.) write to RTT buffer atomically (disable/enable interrupts globally) 
3.) ensure by `Kconfig` that RTT console does not share its RTT up buffer with other subsystems like logger backend (condition for 1.) that uses locking macros and are not usable within isr

Implementation of 2.) could be done super fast as only counting of reservation storage for a message in rtt ring buffer must be done atomically.

pkral78, 2018-11-21:
@carlescufi @nordic-krch comments are welcomed. 

There is also possibility to write to RTT buffer and deal with isr context without disabling interrupts for the entire time of writing to RTT bufer. By introducing atomically incremented pointers to next writable area (which could be done with ldrex/strex) and disabled interrupts only when synchronizing these pointers with offset pointers for RTT buffers. However this requires to implement custom writing functions - not big deal, but this will render official SEGGER API unusable.

So question is, do we really need this? I think not, for this particular issue, the logger redirection solves it elegantly.

MarekPieta, 2018-12-07:
Ok, we may use logger redirection and this solves printk issue.

Anyway we got similiar problem when we want to write to RTT buffer in interrupts for example when we are using profiler (like SystemView). Some time ago there was separate thread for SystemView profiling, but now it uses hooks, so we may write to RTT nearly from any context - it may cause the same problem as it was reported for printk. It may be solved using additional thread, but this would take additional time and for bigger systems it may be serious issue. I think that we could use irq_locks for RTT write instead of using mutex. Writing to RTT does not last long and it shouldn't cause problems.

Changing from mutex to irq_lock would also fix issue:
https://github.com/zephyrproject-rtos/zephyr/issues/11244

pkral78, 2018-12-07:
@MarekPieta Intent not to lock an irqs in locking macros is that default (i.e. channel 0) up/down buffers are shared across many subsystems on Zephyr that simply does not need to lock irqs.

You should allocate and use different RTT channel for SystemView, assigned exclusively for this purpose. I do not know details how SystemView works, but to deal with irq priority there, it of course need to lock irq (or if latence would be unacceptable then some clever locking which suggested in discussion above). I just hope that SystemView does not simply reuses SEGGERS RTT locking macros directly and defines its own (overridable) locking macros. 

For more details see original discusion in #10659

pdunaj, 2018-12-07:
Hi @pkral78 ,
First and foremost there is only one lock in RTT called (`SEGGER_RTT_LOCK`) that is taken whatever destination buffer is selected.
Second, profiler hooks are called from various places inside the system - some of them from interrupt context in which you cannot take mutex.

There are 3 ways of solving issue in profiler:
1. Create a dedicated thread for deferred send of profiler data. Seems obvious but first, we will not avoid taking a lock (irq_lock) at least for a short time and second, deferring the send does not mean there will be no impact on system performance as you need more memory and CPU power to maintain it. Also processing thread cannot have low priority as data will be lost in highly load systems (the exact ones that you usually want to profile).
2. Convert the lock type so we can add data to RTT from any context. This seems expensive as we will hold the lock while filling in the buffer so for a longer time. But this is the simplest possible solution for this system to work. It may also be all-in-all cheaper compared to 1).
3. Rework the RTT subsystem so it would work in lock-less fashion (or nearly lock-less). Note that RTT is already deferred and **there is no point** in making additional thread or buffer. The real problem is that the `SEGGER_RTT_LOCK` is taken for long periods of time. The ideal situation would be for it to be taken only to reserve space in the buffer and when the buffer is mark ready for processing so pointers used by host could be safely updated. Also we should have separate lock for each channel. The problem with this approach is that it will require much more time to implement and we need the profiling now.

pkral78, 2018-12-07:
> Hi @pkral78 ,
> First and foremost there is only one lock in RTT (called `SEGGER_RTT_LOCK`) that is taken whatever destination buffer is selected.

That's just SEGGER's RTT library design limitation. RTT itself does not suffer from such global lock.

> Second, profiler hooks are called from various places inside the system - some of them from interrupt context in which you cannot take mutex.

Right, but it does not implicate that Zephyr should lock whole system because of SystemView.

> There are 3 ways of solving issue in profiler:
>
> 1. Create a dedicated thread for deferred send of profiler data. Seems obvious but first, we will not avoid taking a lock (irq_lock) at least for a [...]
that the `SEGGER_RTT_LOCK` is taken for long periods of time. The ideal situation would be for it to be taken only to reserve space in the buffer and when the buffer is mark ready for processing so pointers used by host could be safely updated. Also we should have separate lock for each channel. The problem with this approach is that it will require much more time to implement and we need the profiling now.

Please do not follow an universal solution approach as it will always cost more but first and foremost is not needed at all. Rule 1. do not increase latency for interrupts if possible. Your (SystemView) subsystem need write to RTT from interrupt context. OK then redirect it to new non-shared channel that are not touched by mutex and use IRQ locking to handle IRQ priority switching. 

If SystemView uses same `SEGGET_RTT_LOCK` macros it just indicate its bad design. Rename it to `SYSTEMVIEW_RTT_LOCK` whatever and use irq_enable/disable .

pkral78, 2018-12-10:
@pdunaj Sorry for late reply. Yes please create new issue and link this to stay aligned with latest discussion.
I'm not against better locking approach, but keep in mind that:
a) current Zephyr RTT consumers (subsystems) do not need to disable interrupts due fact they do not access RTT from interrupt context
b) any better locking will result making RTT library obsolete - see my comment https://github.com/zephyrproject-rtos/zephyr/issues/11393#issuecomment-440572495 - reservation locking you also are talking about
c) due b) it certainly will need to refactor SystemView

PR #14465 (2019-03-19): logging: Force LOG_PRINTK option if RTT backend is used
Log RTT backed is using mutex for locking access to RTT data.
RTT console (which by default is used by printk) writes to
RTT data directly and it cannot use mutex because it can be
called from any context (including interrupt). If both
modules access RTT buffer 0, data can be corrupted.

This patch forces LOG_PRINTK option if RTT backend is used to
ensure single point of access to RTT buffer 0 data.

Fixes #11393.

I agree with points in the the discussion in #11393:
> Well, I don't see CONFIG_LOG_PRINTK as a workaround rather than proper [... cut]
closed 2019-03-19 by galak