zephyrproject-rtos/zephyr #11409: Device crash on BLE reconnection

pdunaj, 2018-11-15:
**Describe the bug**
Device crashes when host reconnects.

**To Reproduce**
Steps to reproduce the behavior:
Connect HID peripheral device to mobile (Android 8). Observe HID activity.
Turn bluetooth off on the host side. Do not restart peripheral. Re-enable BT on host and wait for re-connection. Observe crash.

**Expected behavior**
Crash should not happen.

**Impact**
BLE is unusable.

**Screenshots or console output**
```
[01206140] <err> bt_gatt.bt_gatt_store_ccc: Failed to store CCCs (err -2)
***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x20004f10
Faulting instruction address = 0x0
Fatal fault in thread 0x20004f10! Aborting.
```

**Environment (please complete the following information):**
`f403bfa3434b3802189b3a98905c067828939171`

**Additional context**
N/A

dhananjaygj, 2018-11-15:
@pdunaj We also observed the same issue during our development with windows 10 machine and my colleague @Olivier-ProGlove has created two PRs to solve this issue. 
https://github.com/zephyrproject-rtos/zephyr/pull/11330, https://github.com/zephyrproject-rtos/zephyr/pull/11349.

Hope this solves your problem

Vudentz, 2018-11-21:
Though the settings_save_one does not seems to be the place where it is crashing the error -2 is ENOENT which means something is not configured properly. Does CONFIG_BT_GATT_CLIENT is enabled?

Vudentz, 2018-11-21:
It is known that set callback can be called multiple times with the same setting and I assume that is by design otherwise one will have to have a cache which will retain just the last value found before you commit to each backend, because of this I think it is better that we check if a cfg for the address exists.

Vudentz, 2018-11-21:
@Olivier-ProGlove btw, settings_line_make already support passing NULL as value, that is in fact how keys are cleared here:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/keys.c#L221

Olivier-ProGlove, 2018-11-21:
@Vudentz It is actually the code I was thinking when I mentioned removing setting. When using FCB backend, calling `settings_save_one(key, NULL)` will create a new entry in FCB with an empty value.

When you will load the bluetooth settings, you should expect to load the old key value and the `NULL` key value.

PR #11578 (2018-11-21): Bluetooth: GATT: Fix not handling duplicated CCC storage
Some backends may actually contain the same key multiple times so the
code needs to check if there is already a ccc_cfg for an address before
attempting to use one that is unallocated.

Fixes #11409
commit a618ea2a (2018-11-21): Bluetooth: GATT: Fix not handling duplicated CCC storage

Some backends may actually contain the same key multiple times so the
code needs to check if there is already a ccc_cfg for an address before
attempting to use one that is unallocated.

Fixes #11409
closed 2018-11-21 by jhedberg