zephyrproject-rtos/zephyr #11417: logging doesn't log, again

pfalcon, 2018-11-15:
"Again" is re: https://github.com/zephyrproject-rtos/zephyr/issues/11008

**Describe the bug**
Enabling CONFIG_NET_IPV4_LOG_LEVEL_DBG=y (likely also other similar loggers, at DEBUG level) doesn't lead to LOG_DBG() messages to be logged.

**To Reproduce**
Apply this patch:
~~~
diff --git a/samples/net/echo_server/prj.conf b/samples/net/echo_server/prj.conf
index be16712086..03d8a55797 100644
--- a/samples/net/echo_server/prj.conf
+++ b/samples/net/echo_server/prj.conf
@@ -41,3 +41,5 @@ CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
 CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
 CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
 CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
+
+CONFIG_NET_IPV4_LOG_LEVEL_DBG=y
diff --git a/subsys/net/ip/ipv4.c b/subsys/net/ip/ipv4.c
index 6ab462d84c..64de5fe1d2 100644
--- a/subsys/net/ip/ipv4.c
+++ b/subsys/net/ip/ipv4.c
@@ -116,6 +116,10 @@ enum net_verdict net_ipv4_process_pkt(struct net_pkt *pkt)
        int real_len = net_pkt_get_len(pkt);
        int pkt_len = ntohs(hdr->len);
        enum net_verdict verdict = NET_DROP;
+LOG_INF("net_ipv4_process_pkt");
+NET_INFO("net_ipv4_process_pkt");
+LOG_DBG("net_ipv4_process_pkt");
+NET_DBG("net_ipv4_process_pkt");
 
        if (real_len != pkt_len) {
                NET_DBG("IPv4 packet size %d pkt len %d", pkt_len, real_len);
~~~

Alternatively, it's available in branch https://github.com/pfalcon/zephyr/tree/logging-hosed-again . 

Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=qemu_x86
3. make run (with usual SLIP networking setup for qemu_x86)
4. Then ping from host: ping 192.0.2.1

**Expected behavior**
What I get is:
~~~
uart:~$ ***** Booting Zephyr OS zephyr-v1.13.0-1930-gf60b580710 *****
[00:01:44.607,298] <inf> net_ipv4.net_ipv4_process_pkt: net_ipv4_process_pkt
[00:01:44.607,921] <inf> net_ipv4.net_ipv4_process_pkt: net_ipv4_process_pkt
~~~

Expected is to to see 2 `<dbg>` messages in addition to that.

**Impact**
Logging cannot be relied upon. (Spent more than hour debugging what may be wrong with my input packet that they go thru network in such a path that they got dropped completely silently (indeed, we don't log all possible cases for drops), just in the end got that the logging doesn't really work.)

nordic-krch, 2018-11-15:
Try `log status` in shell. By default shell starts with INF level on all modules. If so enable DBG level for given module. Its an open item how if that level should be configurable. Actually, i discussed same issue today with Robert Lubos.

pfalcon, 2018-11-15:
Thanks for quick response.

But this is very, very wrong. It's nice that it's possible to gate logging level via some runtime-configurable level (I'd personally prefer such feature off by default to save memory). But if user enabled DEBUG level for some logger, they should get their messages. That means that this "current" value should be by default "dbg" instead of "inf".

PR #11434 (2018-11-19): shell: Add initial log level for rtt and uart backends
RTT and uart shell backends were started with fixed
log limit set to INF. Configuration has been moved
to Kconfig allowing certain level or default LOG_MAX_LEVEL.

Fixes #11417.
closed 2018-11-19 by carlescufi