zephyrproject-rtos/zephyr #11447: Build error for lwm2m_client w/ modem overlay

mike-scott, 2018-11-16:
** Issue reproduction **
Using SDK 0.9.5 on Ubuntu 18.04:
```
$ cd <zephyr dir>
$ export ZEPHYR_SDK_INSTALL_DIR=~/zephyr-sdk
$ export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
$ . zephyr-env.sh
$ cd samples/net/lwm2m_client/
$ mkdir build && cd build
$ cmake -DBOARD=frdm_k64f -DOVERLAY_CONFIG=overlay-dns.conf ..
$ make
```

Generates the following build errors:
```
[ 47%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/modem/wncm14a2a.c.obj
/zephyr/samples/net/lwm2m_client/build/zephyr/include/generated/generated_dts_board.h:823:53: error: ‘DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_CONTROLLER’ undeclared here (not in a function)
 #define DT_WNCM14A2A_GPIO_MDM_BOOT_MODE_SEL_NAME    DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_CONTROLLER
                                                     ^
/zephyr/drivers/modem/wncm14a2a.c:46:14: note: in definition of macro ‘PINCONFIG’
  .dev_name = name_, \
              ^~~~~
/zephyr/drivers/modem/wncm14a2a.c:65:12: note: in expansion of macro ‘DT_WNCM14A2A_GPIO_MDM_BOOT_MODE_SEL_NAME’
  PINCONFIG(DT_WNCM14A2A_GPIO_MDM_BOOT_MODE_SEL_NAME,
            ^
/zephyr/samples/net/lwm2m_client/build/zephyr/include/generated/generated_dts_board.h:824:53: error: ‘DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_PIN’ undeclared here (not in a function)
 #define DT_WNCM14A2A_GPIO_MDM_BOOT_MODE_SEL_PIN     DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_PIN
                                                     ^
/zephyr/drivers/modem/wncm14a2a.c:47:9: note: in definition of macro ‘PINCONFIG’
  .pin = pin_ \
         ^~~~
/zephyr/drivers/modem/wncm14a2a.c:66:5: note: in expansion of macro ‘DT_WNCM14A2A_GPIO_MDM_BOOT_MODE_SEL_PIN’
     DT_WNCM14A2A_GPIO_MDM_BOOT_MODE_SEL_PIN),
     ^
/zephyr/samples/net/lwm2m_client/build/zephyr/include/generated/generated_dts_board.h:825:53: error: ‘DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_POWER_GPIOS_CONTROLLER’ undeclared here (not in a function)
 #define DT_WNCM14A2A_GPIO_MDM_POWER_NAME            DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_POWER_GPIOS_CONTROLLER
                                                     ^
/zephyr/drivers/modem/wncm14a2a.c:46:14: note: in definition of macro ‘PINCONFIG’
  .dev_name = name_, \
              ^~~~~
/zephyr/drivers/modem/wncm14a2a.c:69:12: note: in expansion of macro ‘DT_WNCM14A2A_GPIO_MDM_POWER_NAME’
  PINCONFIG(DT_WNCM14A2A_GPIO_MDM_POWER_NAME,
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/zephyr/samples/net/lwm2m_client/build/zephyr/include/generated/generated_dts_board.h:826:53: error: ‘DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_POWER_GPIOS_PIN’ undeclared here (not in a function)
 #define DT_WNCM14A2A_GPIO_MDM_POWER_PIN             DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_POWER_GPIOS_PIN
                                                     ^
/zephyr/drivers/modem/wncm14a2a.c:47:9: note: in definition of macro ‘PINCONFIG’
  .pin = pin_ \
         ^~~~
/zephyr/drivers/modem/wncm14a2a.c:70:5: note: in expansion of macro ‘DT_WNCM14A2A_GPIO_MDM_POWER_PIN’
     DT_WNCM14A2A_GPIO_MDM_POWER_PIN),
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/zephyr/samples/net/lwm2m_client/build/zephyr/include/generated/generated_dts_board.h:827:53: error: ‘DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_CONTROLLER’ undeclared here (not in a function)
 #define DT_WNCM14A2A_GPIO_MDM_KEEP_AWAKE_NAME       DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_CONTROLLER
                                                     ^
/zephyr/drivers/modem/wncm14a2a.c:46:14: note: in definition of macro ‘PINCONFIG’
  .dev_name = name_, \
              ^~~~~
/zephyr/drivers/modem/wncm14a2a.c:73:12: note: in expansion of macro ‘DT_WNCM14A2A_GPIO_MDM_KEEP_AWAKE_NAME’
  PINCONFIG(DT_WNCM14A2A_GPIO_MDM_KEEP_AWAKE_NAME,
            ^
/zephyr/samples/net/lwm2m_client/build/zephyr/include/generated/generated_dts_board.h:828:53: error: ‘DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_PIN’ undeclared here (not in a function)
 #define DT_WNCM14A2A_GPIO_MDM_KEEP_AWAKE_PIN        DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_PIN
                                                     ^
/zephyr/drivers/modem/wncm14a2a.c:47:9: note: in definition of macro ‘PINCONFIG’
  .pin = pin_ \
         ^~~~
/zephyr/drivers/modem/wncm14a2a.c:74:5: note: in expansion of macro ‘DT_WNCM14A2A_GPIO_MDM_KEEP_AWAKE_PIN’
     DT_WNCM14A2A_GPIO_MDM_KEEP_AWAKE_PIN),
     ^
[... 115 lines cut ...]
```
/* wncm14a2a */
#define DT_NXP_KINETIS_UART_4006C000_WNCM14A2A_BUS_NAME					"UART_2"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_LABEL					"wncm14a2a"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_CONTROLLER		"GPIO_2"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_FLAGS		0
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_BOOT_MODE_SEL_GPIOS_PIN			17
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_CONTROLLER		"GPIO_2"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_FLAGS			0
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_KEEP_AWAKE_GPIOS_PIN			2
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_POWER_GPIOS_CONTROLLER			"GPIO_1"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_POWER_GPIOS_FLAGS			0
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_POWER_GPIOS_PIN				9
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_RESET_GPIOS_CONTROLLER			"GPIO_2"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_RESET_GPIOS_FLAGS			0
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_RESET_GPIOS_PIN				12
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_SEND_OK_GPIOS_CONTROLLER		"GPIO_3"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_SEND_OK_GPIOS_FLAGS			0
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_SEND_OK_GPIOS_PIN			0
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_SHLD_TRANS_ENA_GPIOS_CONTROLLER		"GPIO_2"
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_SHLD_TRANS_ENA_GPIOS_FLAGS		0
#define NXP_KINETIS_UART_4006C000_WNCM14A2A_MDM_SHLD_TRANS_ENA_GPIOS_PIN		4
```

This is seems to be an error in the DTS processing?

mike-scott, 2018-11-16:
So it looks like the use of "base_label" in the dt-binding stops the addition of DT_* to the property:
```
$ fgrep -nr "base_label" dts/bindings/
dts/bindings/mtd/partition.yaml:40:base_label: FLASH_AREA
dts/bindings/arm/st,stm32-ccm.yaml:22:base_label: CCM
dts/bindings/modem/wnc,m14a2a.yaml:54:base_label: WNCM14A2A
dts/bindings/interrupt-controller/riscv,plic0.yaml:33:base_label: PLIC
dts/bindings/interrupt-controller/pulpino,event-unit.yaml:27:base_label: PULPINO_EVENT_UNIT
dts/bindings/device_node.yaml.template:73:base_label: LABEL
```

I have to wonder why these are being skipped?

mike-scott, 2018-11-16:
I removed the base_label from the modem binding and adjusted the dts_fixup files in new patches here:
https://github.com/zephyrproject-rtos/zephyr/pull/11448

galak, 2018-11-20:
Is this now resolved with #11448 being merged?

mike-scott, 2018-11-20:
Yep, can close.

closed 2018-11-20 by mike-scott