zephyrproject-rtos/zephyr #11457: change SPI async interface to use callbacks instead of poll

vlotech, 2018-11-17:
The current implementation of spi_transceive_async uses the k_poll_signal mechanism for informing the upper layer the asynchronous transfer is done. This forces the user of this spi interface to use or k_poll and become blocking again (with timeout) or need to poll the signal. 

As most drivers with asynchronous behavior have (in zephyr but also other platforms) is to use a callback function. The user can then choose to the k_poll mechanism or any other mechanism which fits the application architecture. A default callback function which raises the poll_signal (which would be the argument) can be added to allow proper backwards compatibility. 

The alternative is to have a separate thread doing the k_poll part, but that gives additional overhead asuming there is no thread yet who can do this task (e.g. in a event driven application)

The k_poll mechanism can also be extended with a callback method and/or be complemented with a default callback_handler function to be filled in as being the callback pointer for some of the drivers.

vlotech, 2018-11-19:
I'm afraid I disagree with your statement. 

First of all, (almost) all other I/O drivers use callback functions for the async behavior, except the SPI driver. 
Secondly, I think you misunderstood my proposal. It's not the plan to put extensive processing of the SPI data in this callback, but just give the user the freedom to choose it's own way of informing the upper application layer. This could be a poll signal as it is now, but perhaps a semaphore or event on a workqueue or an alert. The poll mechanism has a major disadvantage as it's limited in usage. 
So to sum up. it's not in line with the generic design pattern for drivers and it limits the user to use the default kernel methods to handle synchronization of threads and processes. 

I was already finishing off the SPI driver side of it and I'll port the different drivers myself for this and commit it.

tbursztyka, 2018-11-19:
> First of all, (almost) all other I/O drivers use callback functions for the async behavior, except the SPI driver.

It's true some APIs uses direct cb, but it's because of their domain of use: gpio is commonly used as "interrupts" in many drivers/kernel space, where you need to react locally quickly (and where obviously the cb is then super short), same for watchdog, i2c slaves (there is no async mode for master), same again for IPM, for counter as well, etc... In the end the more complex domains do no have this (besides uart but there is already work being done to get rid of this). Some have it and it's really misplaced there (gna API for instance). So cb are not being used there to provide freedom of choice in informing upper layers or a supposedly generic design pattern, but because of the specific use cases of the domain there are used in. And there is no use case in SPI to provide a low level ISR mode cb, since the only necessity is to signal the end of a transaction (whether it's a success or not).

The issue of having a cb in SPI is that it pushes a common problem to the user to be solved. And that's bad. Thus why it was chosen to go with k_poll_signal. Now, if k_poll_signal is an issue, that's different.

If really changing the event type would be a requirement (and by this, I mean real use-cases behind), a composite type as the async parameter could be a better solution then (a struct with a type, and a union on k_poll_signal, semaphore, fifo, etc... well types that could go in k_poll_event in fact).

vlotech, 2018-11-19:
I agree the responsibility for correct usage moves to the user which can be a problem. 
Therefore I was adding an example cb function for the poll, and can extend it to other examples like semaphore. 

Our use-case is communication with a RF transceiver within a high throughput network, where several SPI calls are needed for the actual transmission and reception to happen. 
Some calls are pretty short (few bytes and therefore not a problem to wait in a blocking manner), and some are larger (e.g. packets sizes up to 1024 bytes). 
In those cases, it's good to free up the CPU for other tasks, but after the transfer, the higher level driver needs to continue asap depending on the state. 
To prevent an additional driver thread for this, k_poll would not be the ideal solution. 

Is there already an implementation using a composite type for asynchronous feedback? if so, I can implement this for SPI, or else I could make a proposal for it (with only event generators, like mutex unlock, semaphore give, etc. etc. and no data feed-forward involved).

tbursztyka, 2018-11-19:
No existing implementation no, feel free to make a proposal.

I think k_poll_signal and k_sem would be the types to support (all generically handled in drivers/spi/spi_context.h). I don't see any point for k_fifo for instance, nor k_queue as spi would have nothing to put there, unless I am missing something here. Not sure about k_work either.

vlotech, 2018-11-19:
Ok, I'll make a proposal and send the url in github.

vlotech, 2018-11-19:
see: https://github.com/vlotech/zephyr/tree/spi_composite_event
It currently contains the driver + interface changes only. 
https://github.com/vlotech/zephyr/commit/b72fbdc0d1b4fdc58cee9974bfeecfe1db90b27f

If this is in line with the preferred solution, I'll go over the drivers making use of the SPI async interface and update those + create a pull request.

tbursztyka, 2018-11-19:
seems like a good start.
Just about naming:
s/SPI_ASYNC_EVENT_/SPI_ASYNC_EVENT_TYPE_
s/POLL/SIGNAL
s/spi_async_config/spi_async_event

and void object, instead make: 
union {
        void *object;
        struct k_poll_signal *signal;
        struct k_sem *sem;
        ... etc ...
};

also the comment: "type of pointer used for the async callback"
better "Type of object used for asynchronous signalling"

tbursztyka, 2018-11-19:
As a PR, create it as an [RFC], I'll link then this PR to relevant people since it requires other relevant people to judge the change.

I believe the change should not be that hard on driver side, mostly a signature change. Biggest changes will be in drivers/spi/spi_context.h

vlotech, 2018-11-19:
[RFC] drivers/spi: implement a composite event for async callback #11500 pull request is created.

pabigot, 2019-04-27:
I like this proposal if it's generalized beyond SPI to ADC, and possibly other async-able subsystems like sensors.

I agree that assuming `k_poll` with a signal is always suitable makes the ADC (and SPI) async APIs difficult to use in some cases.  So I'd like to see a generic way to specify how asynchronous results are conveyed to the caller.  (I personally prefer a callback as it's the most flexible, but the arguments against it above have some weight, so I won't fight for that right now.)

If #15703 leads to a rework of how the SPI async context is managed then adding this flexibility at the same time should be considered.  That rework should be designed support any other async operation for which an `int` status is provided.  It would require that the object that captures the notification method and resources and the status would have to be in an object specific to the operation, owned by the caller, probably extended with driver-specific information (SPI buffers and sizes; ADC configurations).  So the approach in #11500 would have to be reworked significantly.

pabigot, 2020-08-25:
Just to update: [async_notify](https://docs.zephyrproject.org/latest/reference/misc/notify.html#async-notification) is the current in-tree solution for configuring notification methods.