zephyrproject-rtos/zephyr #11461: tests: intel_s1000: build error due to offsets lib

sathishkuttan, 2018-11-18:
**Describe the bug**
A commit (18f9bb0) in master is causing build failure for intel_s1000

**To Reproduce**
Rebase to commit 18f9bb0
Steps to reproduce the behavior:
In `tests/boards/intel_s1000_crb`
```
mkdir build; cd build
cmake -DBOARD=intel_s1000_crb ..
make
```
See error

**Expected behavior**
Build should succeed without errors.

**Impact**
Impacts work on #11390

**Console output**
```
Scanning dependencies of target offsets_h
[  8%] Generating include/generated/offsets.h
usage: gen_offset_header.py [-h] -i INPUT -o OUTPUT
gen_offset_header.py: error: argument -i/--input: expected one argument
zephyr/CMakeFiles/offsets_h.dir/build.make:60: recipe for target 'zephyr/include/generated/offsets.h' failed
make[2]: *** [zephyr/include/generated/offsets.h] Error 2
CMakeFiles/Makefile2:551: recipe for target 'zephyr/CMakeFiles/offsets_h.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/offsets_h.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

nashif, 2018-11-18:
which cmake version are you using?

`cmake --version`

sathishkuttan, 2018-11-18:
> which cmake version are you using?

cmake version 3.8.2

`tests/boards/intel_s1000_crb/CMakeLists.txt` has
`cmake_minimum_required(VERSION 3.8.2)`

nashif, 2018-11-18:
ok, I confirm this does not work with 3.8.2, looks like most of us are using newer versions.

Will revert the patch until we have found a way around this or updated minimum cmake requirement.

@kapedev @SebastianBoe fyi