zephyrproject-rtos/zephyr #11490: setup_ipv6() treats event enums as bitmasks

zagor, 2018-11-19:
Here is a snippet from `subsys/net/lib/config/init.c`. I've added a comment in brackets after each assignment of `mask`, showing the result on the 'cmd' bits of mask:

```C
static void setup_ipv6(struct net_if *iface, u32_t flags)
{
	struct net_if_addr *ifaddr;
	u32_t mask = NET_EVENT_IPV6_DAD_SUCCEED; // [mask = 0b1101]

	if (sizeof(CONFIG_NET_CONFIG_MY_IPV6_ADDR) == 1) {
		/* Empty address, skip setting ANY address in this case */
		return;
	}

	if (net_addr_pton(AF_INET6, CONFIG_NET_CONFIG_MY_IPV6_ADDR, &laddr)) {
		NET_ERR("Invalid address: %s", CONFIG_NET_CONFIG_MY_IPV6_ADDR);
		/* some interfaces may add IP address later */
		mask |= NET_EVENT_IPV6_ADDR_ADD; // [0b1101 | 0b0001 = 0b1101]
	}

	if (flags & NET_CONFIG_NEED_ROUTER) {
		mask |= NET_EVENT_IPV6_ROUTER_ADD; // [0b1101 | 0b1011 = 0b1111]
	}
```

After the statement `mask |= NET_EVENT_IPV6_ADDR_ADD`, the value of mask is unchanged, so is still `NET_EVENT_IPV6_DAD_SUCCEED`. The statement is thus a no-op.

After the statement `mask |= NET_EVENT_IPV6_ROUTER_ADD`, mask has the rather surprising value
`NET_EVENT_IPV6_NBR_ADD`.

This seems very wrong to me.

(This report has similarities to https://github.com/zephyrproject-rtos/zephyr/issues/6413.)

Ping @jukkar.

tbursztyka, 2018-11-19:
It's by design, the "mask" isn't a strict bit mask. At that time there was a big urge to save as much memory as possible, and thus a combination of events can make events to overlap, it was considered less harmful to do so as it would just call the callback some time too many, instead of eating up memory on each and every event registered, knowing that these evens are anyway seldom. It's just up to the event callback to verify the event as it always gets it.

zagor, 2018-11-19:
I don't see how it calls the event handler too many times, I see it using the wrong value. The calling code tries to set DAD_SUCCEED and ROUTER_ADD, but ends up sending NBR_ADD. There is no way for the event handler to find out that the caller tried to do something other than NBR_ADD.

And ipv6_event_handler() never tries to decipher the mask, it always compares full integer value:

```C
static void ipv6_event_handler(struct net_mgmt_event_callback *cb,
			       u32_t mgmt_event, struct net_if *iface)
{
	struct net_if_ipv6 *ipv6 = iface->config.ip.ipv6;
	int i;

	if (!ipv6) {
		return;
	}

	if (mgmt_event == NET_EVENT_IPV6_ADDR_ADD) {
		/* save the last added IP address for this interface */
		for (i = NET_IF_MAX_IPV6_ADDR - 1; i >= 0; i--) {
			if (ipv6->unicast[i].is_used) {
				memcpy(&laddr,
				       &ipv6->unicast[i].address.in6_addr,
				       sizeof(laddr));
			}
		}
	}

	if (mgmt_event == NET_EVENT_IPV6_DAD_SUCCEED) {
#if NET_LOG_LEVEL > 2
		char hr_addr[NET_IPV6_ADDR_LEN];
#endif
	...
```

If this is by design then it needs to be thoroughly documented. I'm not going to be the last person looking at this and saying "this can't be right".

jukkar, 2020-01-28:
Going through old issues. This is a good candidate for closing as there currently there is no plan to change how the net-mgmt works. Can be re-opened if really needed.

closed 2020-01-28 by jukkar