zephyrproject-rtos/zephyr #11495: Printk overwriting each other when processed

Qbicz, 2018-11-19:
**Describe the bug**
Enabled processing of printk messages by logger: `LOG_PRINTK=y`, with RTT backend, two subsequent printk calls without line break. Effect: second message is printed from beginning of the line and overwrites the first printed message.

**To Reproduce**
Steps to reproduce the behavior:
1. `LOG_PRINTK=y`
2. Use RTT backend to log
3. `printk("1: %s ", "first_printk"); printk("2: %d ", 7); printk("\n");`
4. `2: 7rst_printk`

**Expected behavior**
Printouts should come one after another.

**Impact**
Incorrect printing of logs, medium priority problem.

**Environment (please complete the following information):**
 - OS: Ubuntu 16.04 LTS
 - Toolchain: Zephyr SDK 0.9.5
 - Commit SHA or Version used: 9c14767475a68
 - HW: nRF52840

Qbicz, 2018-11-19:
FYI @nordic-krch @pdunaj

closed 2018-11-23 by nashif

pdunaj, 2018-11-23:
It is possible that this bug is not yet completely fixed. Please check the comment in the PR.

PR #11598 (2018-11-23): logging: Fix lost printk output when no newline
Log_output module was always postfixing printk strings
with '\r'. If printk message did not ended with '\n'
it lead to last printk message being overwritten.

Fixed by adding '\r' only when string was ended by '\n'.

Fixes #11495,

Note that output will not look nice if consecutive printks (without '\n' in between) will
be interleaved with log messages. 

Example:
```
printk("abcd")
printk("1234\n");
```

Expected output:
```
abcd1234
[12232132] <DBG> .....
```

Possible output
```
abcd[12321321] <DBG> ...
1234
```

In order to fix that log_outpu [... cut]