zephyrproject-rtos/zephyr #11501: RFC: Improve CI and add more status items

nashif, 2018-11-19:
## Current Status

Currently all of our CI is run in one job and all results are reported via one status from shippable. We often have to wait more than 30 minutes to get a report that only complains about bad commit messages and such. Thos reports are hidden in  the over all report and not immediately visible to new comers and first time contributors.

## Proposal

- Move the compliance tests to a separate job that runs in parallel and reports to Github individually per test case (compliance and documentation tests).

- Report the output of the test to GH directly as a comment, this is visible to the authors directly, no need to go search for the report on shippable

- Each test has a status with a link to the guidelines

[image]

- All the compliance tests are run with ~10 minutes and reported to the PR as a comment before the actual sanitycheck run is done, giving developer immediate feedback.

## Things to consider

- Currently the recheck command in comments will only work with sanitycheck and full CI run, not the compliance statuses

- Documentation build is a bit slow, so the compliance job will take approx 10 min, this needs to be optimized in the future

- Documentation build is moved out of the main job into the new one, leaving the main CI job only for running sanitycheck.

galak, 2018-11-19:
Can we add recheck-{checkpatch, doc, gitlint, etc.., all}?

ceolin, 2018-11-19:
Can we add soft checks ? We have some coccinelle scripts that would be
great to be checked. Some of them are testing MISRA-C rules and it'd
nice to have a report. The problem is that they run per file/directory
so we can have issues not related with the patch, also it can have
false-positives. Though, this still can be used to help on review.

galak, 2018-11-19:
> Can we add soft checks ? We have some coccinelle scripts that would be
> great to be checked. Some of them are testing MISRA-C rules and it'd
> nice to have a report. The problem is that they run per [...]

I wondering about using something like https://lgtm.com/ for such checks.

nashif, 2018-11-20:
> Can we add recheck-{checkpatch, doc, gitlint, etc.., all}?

i would prefer to have 1 single recheck that does everything and that will be done but not immediately, there is no value on doing them individually, it will take almost exactly the same time.

nashif, 2018-11-20:
> Can we add soft checks ? We have some coccinelle scripts that would be
> great to be checked. Some of them are testing MISRA-C rules and it'd
> nice to have a report. The problem is that they run per [...]

yes, this can be done without reporting a status, this can probably just generate a report and post it somewhere for review. We can always run this on master and have a baseline and compare the results of the PR with master and report the diff for example.

nashif, 2019-11-21:
done already, so closing.

closed 2019-11-21 by nashif