zephyrproject-rtos/zephyr #11502: Delayed works are not scheduled when system is busy

pdunaj, 2018-11-19:
**Describe the bug**
Replicated with nrf52840, revision
We execute most of our things from workqueue thread via works.
During the latest testing we found out that when there is a lot of normal works, delayed works are not scheduled for execution.
Quick investigation shows that delayed work is never submitted to the workqueue until system has nothing else to do. I.e. when user stops generating hid events system finally gets time to check expired delayed works and submit them.

Another quick look into the timeout shows that when problem happens `z_clock_announce` is not called. 

Note that normal works are executed without a problem. Also low priority thread is executed without a problem while issue is seen.

**To Reproduce**
Steps to reproduce the behavior:
Enable USB, periodically transmit HID reports via USB through normal works.
Schedule delayed work that would periodically run some check.
Observe that when when frequency of normal work calls increases at some point delayed work is not scheduled.

**Expected behavior**
Delayed work should always be scheduled when expired if there is CPU time to handle normal work (i.e. workq thread is normally scheduled).

**Impact**
showstopper

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
rev:`9c14767475a682c20927746dc652a0e644922241`

pdunaj, 2018-11-19:
@pizi-nordic

pdunaj, 2018-11-19:
it looks as if `rtc1_nrf5_isr` is not called when issue is seen

... and the at least once was randomly gone after some time... it could be that device went through the low power cycle but I am not sure and could not replicate it so far.

pdunaj, 2018-11-20:
I am looking at the code. There seems to be a bug in `_add_timeout` from `kernel/timeout.c`.
Note that `z_clock_set_timeout(_get_next_timeout_expiry(), false);` is called without a lock held.

When two contexts are calling `_add_timeout` add the same time `z_clock_set_timeout` could set the wrong timeout. E.g.
Low priority context calls `_add_timeout` with long timeout. Timeout is added to the list, lock is released and `_get_next_timeout_expiry()` is called to calculate the amount of ticks to wait.
At this moment (before the call to `z_clock_set_timeout`) high priority context grabs the CPU, adds its short timeout to the list, lock is released, next expiry is calculated and timeout set. So far it's ok but now CPU is given back to the low priority context that continues with a call to `z_clock_set_timeout`. It uses the local value that has long timeout. This call will override the short timeout set by high priority context!

I am not sure this is the cause but this is clearly a bug.
@pizi-nordic

carlescufi, 2018-11-20:
Setting this as priority high tentatively for now.

andyross, 2018-11-20:
Yeah, that indeed looks like a race in the timeout handling.  Will audit and fix.

pdunaj, 2018-11-20:
It seems that locks do not work...

```
void z_clock_announce(s32_t ticks)                                                                                       
{                                                                                                                        
        struct _timeout *t = NULL;                                                                                       
                                                                                                                         
#ifdef CONFIG_TIMESLICING                                                                                                
        z_time_slice(ticks);                                                                                             
#endif                                                                                                                   
                                                                                                                         
        //LOG_ERR("");                                                                                                   
        LOG_ERR("%u", _EXC_IRQ_DEFAULT_PRIO);                                                                            
        announce_remaining = ticks;                                                                                      
        while (true) {                                                                                                   
                LOCKED(&timeout_lock) {                                                                                  
                        t = first();                                                                                     
                        if (t != NULL) {                                                                                 
                                static unsigned marker = (unsigned)(-1);                                                 
                                marker--;                                                                                
                                LOG_ERR("%d %d %u", t->dticks, announce_remaining, marker);                              
                                if (t->dticks <= announce_remaining) {                                                   
                                        announce_remaining -= t->dticks;                                                 
                                        curr_tick += t->dticks;                                                          
                                        t->dticks = 0;                                                                   
                                        remove_timeout(t, marker);
```

```
int _abort_timeout(struct _timeout *to)                                                                                  
{                                                                                                                        
        int ret = _INACTIVE;                                                                                             
        static unsigned count;                                                                                           
                                                                                                                         
        LOCKED(&timeout_lock) {                                                                                          
                LOG_ERR("a");                                                                                            
                if (to->dticks != _INACTIVE) {                                                                           
                        remove_timeout(to, count++);                                                                     
                        ret = 0;                                                                                         
                }                                                                                                        
        }    
```

```
static void remove_timeout(struct _timeout *t, unsigned place)                                                           
{                                                                                                                        
        static unsigned count;                                                                                           
        LOG_ERR("%p %u %u", t, count++, place);                                                                          
        if (next(t) != NULL) {                                                                                           
                next(t)->dticks += t->dticks;                                                                            
        }                                                                                                                
                                                                                                                         
        sys_dlist_remove(&t->node);                                                                                      
        t->dticks = _INACTIVE;                                                                                             
}
```

and log:
```
[00042593] <err> dupa.z_clock_announce: 997 0 4294966882
[00042594] <err> dupa._abort_timeout: a
[00042594] <err> dupa.remove_timeout: 0x20003ea0 161 130
[00042595] <err> dupa.remove_timeout: 0x20003ea0 162 4294966882
```

pdunaj, 2018-11-20:
note that `remove_timeout` is called twice from two contexts the problem is should not happen as first call should take it off the list.
Check the marker I added. Observe that `4294966882` is seen in log generated by `_abort_timeout` then we see two removes but before we see the remove from `z_clock_announce` we get another remove called by `_abort_timeout`.
This should not happen as locks are held!

andyross, 2018-11-20:
OK.  I'll let you look at the nRF interrupt handling.  But as described, that definitely sounds like a race in the timeout code, even if the causation of the symptom is more complicated.  I'll start there.

andyross, 2018-11-20:
Patch to address the timeout race (which is very real) is up at #11539

pdunaj, 2018-11-20:
Hi @carlescufi , @anangl , the priorities can be set in the device tree but I think users would expect defaults to be reasonable. It seems however that most peripherals have the same level as RTC clock that is responsible for system scheduling.
Also it would be worth checking if these priorities are correct with regard to zero latency interrupts? (I will check it tomorrow but please tell if you know it from the top of your head.)

carlescufi, 2018-11-20:
> It seems however that most peripherals have the same level as RTC clock that is responsible for system scheduling.

Yes, that is something that we could change so that the RTC gets more time and is not choked by the rest of the peripherals, agreed.

>Also it would be worth checking if these priorities are correct with regard to zero latency interrupts?

ZLI, when enabled, shifts the whole interrupt priority scale, so the same numbers apply, IIRC

andyross, 2018-11-20:
Drive by comment: devicetree isn't really the right place to be assigning interrupt priorities.  Those are software configuration and different apps may want to make different decisions.  An app that does all its work out of timeouts and work queues, like yours, probably wants the timer interrupt to be high priority so that events happen at the right time regardless of hardware activity.  An app that demands low latency to devices so it can respond to real time events out of ISRs wants that interrupt to be higher priority so that it doesn't get delayed by a mere timer interrupt.

andyross, 2018-11-20:
And FWIW: about BASEPRI being 32, that doesn't sound inherently wrong.  I'm not an ARM guy obviously, but BASEPRI is an integer that gets compared, not a mask.  Architecturally I believe it's guaranteed 8 significant bits, so if you have only 8 interrupt levels you obviously don't care about the other 248 values, but it's not wrong to use them.   The interrupt locking implementation is in include/arch/arm/cortex_m/asm_inline_gcc.h.  You can hand-edit it to use a different maximal BASEPRI, or even past in the old style one based on CPSID/CPSIE.

andyross, 2018-11-21:
I couldn't let go of the "locks don't work" so came back to look at this again.

At least on qemu_cortex_m3 (I don't have an nRF board handy to check), the problem is the converse: **interrupt handlers are called with BASEPRI == 0, so interrupts aren't locked**.  Check with a code block like this:

	unsigned int basepri;
	__asm__ volatile("mrs %0, BASEPRI" : "=r"(basepri));
	printk("BASEPRI == %d\n", basepri);

Obviously it's zero in a normal thread, and as mentioned it's 32 when interrupts are locked.  But checking at the top of _timer_int_handler(), it **should** be equal to the IRQ priority (not sure what that is exactly, but on Cortex M3+ we shouldn't be relying on the CPSI bit so we can have nested interrupts) to mask out lower priority interrupts.  And in fact grepping around I don't see anything in ARM interrupt entry that tries to get this right...

Maybe something in the hardware is expected to get this right and isn't enabled?  I'd need to do some reading on ARM interrupt handling to give a good answer, but this for sure

pdunaj, 2018-11-21:
@pizi-nordic checked the basepri value and it is ok (he will add a comment about that).

I have checked from which interrupt two contexts are called and it is the same value (the same vector from IPSR). I have one hypothesis that I want to check. I use LOG_ERR to print the messages - if this call sometimes causes a logging thread to be woken up it would call remove timeout. Since I call LOG_ERR from the same context it would explain why `abort_timeout` is called in the middle of processing of `z_clock_announce` and under the lock. I will check it today but it seems to me that this is the cause.

In the end it may be that problem with locks was caused by improper use of LOG. It may mean that fix provided that @andyross will work when I remove all calls to LOG in the file.

pdunaj, 2018-11-21:
@pizi-nordic , @nordic-krch this is nothing logger can solve. I think the best way would be to add a comment in this file that you cannot call anything that may involve scheduler. This is for future generations to not make the same mistake.

pizi-nordic, 2018-11-21:
The LOG() macros should work from any context (with deferred logging enabled). So they should not interfere timers and other system areas. If this is happening, I consider this as a bug.

pdunaj, 2018-11-21:
The original problem with delayed works is still visible even with the patch applied.
Logs collected suggest that RTC isr is not called. The problem is that I know our preemptive task is running fine (I get correct readouts from the sensor) so I don't expect high priority context to hold the CPU. Everything suggests that timeout set is wrong... but why?

nordic-krch, 2018-11-21:
there is an option in logger `LOG_PROCESS_TRIGGER_THRESHOLD` which wakes up logger thread on given amount of pending log messages (using `k_wakeup`). 0 disables the feature. You can check if that helps. 

>So they should not interfere timers and other system areas. If this is happening, I consider this as a bug.

If there are contexts from which `k_wakeup` call is forbidden that should be documented or fixed.

pdunaj, 2018-11-21:
I have found the problem. It is in `drivers/timer/nrf_rtc_timer.c` and will emanate itself when system experience frequent calls to `z_clock_set_timeout`. This can happen when something in the system has constantly updated timeout (e.g. interrupt cancels and re-enables delayed work). By frequent I mean occurring several times within one tick.
 
The problem is that when you do that code in `z_clock_set_timeout` will move the value of the comparator into the future. There is a check that difference between the comparator value and current counter value is not lower then `MIN_DELAY` (and this is the minimum being set). If during this time something will update the timeout will flow into the future. If you keep updating the timeout the interrupt will not be called as comparator will be moved further and further.
The only way to break it is to allow the comparator to fire.

This issue will occur when we should fire in the current tick or are already delayed (i.e. ticks argument passed to  `z_clock_set_timeout` is zero).

pdunaj, 2018-11-21:
Hi @andyross , I don't think `MIN_DELAY` should be removed. Maybe we should not set the timeout if update of the list did not bring anything new to the front. The change would be then in `_add_timeout` - before calling `z_clock_set_timeout` we could check if front of the queue is different.

I can give it a try now and see if this helps.

pdunaj, 2018-11-21:
@andyross , I gave it a try.
I still think we should add
```
                if (to == first()) {
                        z_clock_set_timeout(_get_next_timeout_expiry(), false);
                }
```
but it is not enough.

The problem is that `z_clock_set_timeout` is also called from `sys_power_save_idle`. And here it is called with tick set to 1 that when late will cause the timeout drift.

Only when I removed
```
        if ((cyc - t) < MIN_DELAY) {
              cyc += CYC_PER_TICK;
        }
```
The problem was gone. So it seems this is needed.
Maybe `idle` should not call `sys_power_save_idle` or this `z_clock_set_timeout` from if ticks are short? This should be reviewed I think.

BTW. Even if we fix the `sys_power_save_idle` there is still the problem when somebody will constantly add and remove timeout with short value (i.e. 1 or 0) as that may caused to to be always placed first on the list and so impact the next element by drifting the timeout counter.

andyross, 2018-11-21:
Ah, OK.   RISC-V actually had the same glitch.  There it was fatal -- the architecture idle implementation doesn't have a "wait for next interrupt", so its behavior was to spin on the idle loop by default, calling z_clock_set_timeout(..., true) in a tight loop, so the timer was always getting "too close" to the expiring tick and bumping it to the next one, and timers would never expire.

The fix there was simply to ignore the call if idle == true.   The OS is guaranteed to have told you about the current timer already, we don't have to do anything special.   Really that idle parameter is legacy -- it used to be that "idle timeout" and "regular timeout" were different, and this is how I folded them together.  Now none of the drivers care about the difference  (Though in theory someone might!  It's possible to imagine a system that uses different hardware for wakeup timers than for runtime ones, and it would need to know that the system was about to idle to decide which one to use.  This is theoretical though, the existing drivers don't care.)

So for now, can you try just ignoring idle==true in the driver?

In the more immediate term we need protection inside the kernel against the "reset the next timeout when we are very near the current one" case, which I suspect can simply be detecting the case where you're calling a timeout with ticks <=0 and still haven't had the z_clock_announce() call for the last one.

andyross, 2018-11-21:
(FWIW, my worry about about interrupts not being locked against each other was a red herring.  Further reading shows that on ARM the NVIC will dispatch nested interrupts on its own and wait for end of service, and that nesting is done internally and not reflected in the state of BASEPRI or PRIMASK.   Well, I learned something anyway.)

andyross, 2018-11-22:
FWIW: I got started on a proper fix for this (which takes the "sleep for N ticks" decision out of the idle loop entirely), but I'm not going to get it submitted tonight.  And note that the rest of the week is a US holiday, though I should be able to find some time.

In the short term, please try the "ignore z_clock_set_timeout() when idle == true" trick in the driver, which should have the same effect.  You can check the RISC-V driver to see the same logic.

pdunaj, 2018-11-22:
Hi @andyross , thanks. I can create a patch if you are busy.
I think the best way to fix it would be to set the counter compactor always through timeout infrastructure. Otherwise we may always get a chance for invalid timeout being set. I will give it a try and create a PR for review.

pdunaj, 2018-11-22:
I added a PR with a fix.