zephyrproject-rtos/zephyr #11548: mimxrt1050_evk: On-chip SRAM appears to be wasted

pfalcon, 2018-11-20:
@agansari complained that he can't run biggish samples. I looked what memmap for this board is.

So, in dts/arm/nxp/nxp_rt.dtsi we have:

~~~
                        itcm0: itcm@0 {
                                reg = <0x00000000 0x20000>;
                        };

                        dtcm0: dtcm@20000000 {
                                reg = <0x20000000 0x20000>;
                        };

                        ocram0: ocram@20200000 {
                                reg = <0x20200000 0x40000>;
                        };
~~~

But in generated_dts_board.h, of memories there's only:

~~~
/* dtcm@20000000 */
#define CONFIG_SRAM_BASE_ADDRESS        0x20000000
#define CONFIG_SRAM_SIZE                128

/* itcm@0 */
#define CONFIG_FLASH_BASE_ADDRESS       0x0
#define CONFIG_FLASH_LOAD_OFFSET        0
#define CONFIG_FLASH_LOAD_SIZE          0
#define CONFIG_FLASH_SIZE               128
~~~

Which is kinda corresponds to what we have in boards/arm/mimxrt1050_evk/mimxrt1050_evk.dts:
~~~
        chosen {
#if defined(CONFIG_CODE_ITCM)
                zephyr,flash = &itcm0;
#elif defined(CONFIG_CODE_HYPERFLASH)
                zephyr,flash = &hyperflash0;
#elif defined(CONFIG_CODE_QSPI)
                zephyr,flash = &qspi0;
#endif
                zephyr,sram = &dtcm0;
~~~

ocram0 is not listed anywhere. So, we effectively throw away 256K of RAM, while limiting text segment to just 128K.

Now, from quick grep, I don't see where flexram controller is initialized, so maybe above is the default post-reset config. But we definitely need to look into changing that.

agansari, 2019-02-04:
@pfalcon can be closed with merge #12465

pfalcon, 2019-02-04:
@agansari:

> @pfalcon can be closed with merge #12465

Well, that's networking PR. Did you mean https://github.com/zephyrproject-rtos/zephyr/pull/12237 instead?

agansari, 2019-02-05:
> Well, that's networking PR. Did you mean #12237 instead?
Both, #12237 enables driver support on all i.mx rt boards, but in #12465's first commit they are set as default on rt 1050.

pfalcon, 2019-02-05:
> Both, #12237 enables driver support on all i.mx rt boards, but in #12465's first commit they are set as default on rt 1050.

Well, that's why a PR titled "networking fixes" shouldn't contain "external flash and sdram" commits, it can only lead to confusion ;-).

I would note that the topic of this issue doesn't see to be addressed directly (adjusting usage ratio of on-chip SRAM), pragmatically it was about "we need more RAM for program storage), and if it's addressed by switching to enormous SDRAM, yeah, let's close this ticket.

closed 2019-02-05 by pfalcon