zephyrproject-rtos/zephyr #11558: Need notification when a BLE packet is sent

m-nilsson, 2018-11-21:
Hi,

We would like to have a notification when a BLE GATT packet is sent.

Our current use-case is to implement accurate clock sync between BLE devices. But I am guessing that there are also many other applications where you want to know if and when a packet is sent.
We have partly solved the problem by calling a hook function from att_pdu_sent() in subsys/bluetooth/host/att.c.
Then we can get a notification when a packet is sent that was previously queued by bt_gatt_write_without_response().

A better solution would be if there was an API where you could add extra parameters to bt_gatt_write_without_response() and also bt_gatt_notify(). And possibly the other sending functions as well.

The extra parameters could be a pointer to a callback function and a void* to some user data.
And then that callback would be called when the transmission of the packet completed.

Or you could register the callbacks once and then only supply the user data when sending the packet.

The problem with the current API is that after requesting to send some data you really have no idea of when sending actually takes place. You only know that it was put in the queue and will be sent some time in the future. But this small change to the API would open up the possibilities to implement more time critical bluetooth applications.

m-nilsson, 2018-11-21:
Update:
I saw that in the master branch there is the addition of the function bt_gatt_notify_cb() that does almost what I requested. So then my feature request is to add a similar function, bt_gatt_write_without_response_cb(). And also to be able to pass some user data to those functions, that is in turn passed to the callbacks. That way it is easier to match a request with a callback, especially when there are several requests in the queue.

m-nilsson, 2018-11-21:
I have a question about the new function bt_gatt_notify_cb().

When I used bt_gatt_write_without_response() I saw that att_pdu_sent() was called about 140us after the packet was received at the peripheral.

However, when using bt_gatt_notify_cb(), the callback is called more than 6ms after the packet is received at the central.
Why this long delay before the callback is executed?

Vudentz, 2018-11-22:
Waiting the controller to acknowledge will always create delays, in fact you don't really need such mechanism if you can block since bt_gatt_write_without_response will do exactly that when attempting to allocate a buffer. HCI already has flow control, in fact that how we are acknowledge that a packet has been sent over the air and free the buffer associated with it which may wakeup any thread waiting for more buffers so in practice it work as a credit based flow control where credits are buffers and when there are exhausted it blocks until more buffer are available.

That said I don't oppose to introduce a similar concept as bt_gatt_notify_cb, the callback would take the parameters given which can then be used with CONTAINER_OF to access outer struct which can contain user_data.

m-nilsson, 2018-11-22:
But bt_gatt_write_without_response will not block if there are buffers available, right?

So what is the difference between bt_gatt_write_without_response and bt_gatt_notify_cb? They don't require response from the other peer and they both end up in bt_l2cap_send_cb, so where is that extra delay coming from? Sorry if it is a dumb question, but I am not familiar with the lower levels of the ble stack.

For precise clock synchronization, it is helpful to have a callback at a reasonable predictable time offset from actual radio transmission, and preferable as close as possible to that.

I'm happy to hear that you are positive to introduce write with callback and user data!

PR #12472 (2019-01-17): Bluetooth: GATT: Add bt_gatt_write_without_rsp_cb
This adds bt_gatt_write_response_cb works similarly to bt_gatt_notify_cb which can take a callback to be called when the PDU is considered transmitted over the air.

Fixes #11558
commit a779705f (2019-01-17): Bluetooth: GATT: Add bt_gatt_write_response_cb

This adds bt_gatt_write_response_cb works similarly to
bt_gatt_notify_cb which can take a callback to be called when the PDU
is considered transmitted over the air.

Note: This can also be used to disable the ATT flow control which would
blocks sending multiple commands without wainting their transmissions.

Fixes #11558
closed 2019-01-17 by nashif