zephyrproject-rtos/zephyr #11561: Ping through net shell seems to be broken

tgorochowik, 2018-11-21:
Running ping through the net shell causes a `BUS FAULT`. Tested on `frdm_k4f` and `sam_e70_xplained`.

One scenario where it happens is using http server with net shell and dhcp enabled, here is the full config:
```
# Generated by Kconfiglib (https://github.com/ulfalizer/Kconfiglib)
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=120000000
CONFIG_SOC_SERIES_KINETIS_K6X=y
CONFIG_ARM_MPU=y
CONFIG_ARM=y
CONFIG_INIT_STACKS=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_PINMUX=y
CONFIG_NETWORKING=y
CONFIG_NET_LOG=y
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_DHCPV4=y
CONFIG_NET_SHELL=y
CONFIG_NET_TCP=y
CONFIG_NET_TCP_BACKLOG_SIZE=2
CONFIG_NET_MAX_CONTEXTS=16
CONFIG_NET_PKT_RX_COUNT=32
CONFIG_NET_PKT_TX_COUNT=32
CONFIG_NET_BUF_RX_COUNT=32
CONFIG_NET_BUF_TX_COUNT=32
CONFIG_NET_STATISTICS=y
CONFIG_HTTP=y
CONFIG_HTTP_SERVER=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_MY_IPV4_NETMASK=""
CONFIG_NET_APP_SERVER_NUM_CONN=2
```

And here is the output:
```
uart:~$ **** Booting Zephyr OS zephyr-v1.13.0-2035-ge85986e153 *****

uart:~$ 
uart:~$ net ping 192.168.0.100
Sent a piSent a ping to 192.168.0.100
Received eG reply from 192.168.0.100 to 192.168.0.102
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0xe3d7ab41
***** Hardware exception *****
Current thread ID = 0x20000fdc
Faulting instruction address = 0x1b9ce
Fatal fault in essential thread! Spinning...
```

Note that the output is also kind of malformed (e.g. `Received eG reply` instead of `Received echo reply`). This happens when pinging a PC connected to the same router, or other Zephyr devices.

@jukkar could you take a look at this?

qianfan-Zhao, 2018-11-21:
please take a look: #11281
could you please try revert 4962618 and try again?

tgorochowik, 2018-11-21:
@qianfan-Zhao thanks! That indeed seems to be the same issue. Reverting that commit is not that straightforward as there are some conflicts, but disabling `CONFIG_UART_INTERRUPT_DRIVEN` (like you recommended in #11281) which effectively disables the ring-buffers fixes the issue for me too.

I'm closing this one as this is a duplicate then, thank you again!

closed 2018-11-21 by tgorochowik