zephyrproject-rtos/zephyr #11564: gatt: settings: Invalid base64 value written to flash

Qbicz, 2018-11-21:
**Describe the bug**
When central connects, but does not subscribe to any notification, an invalid base64 value is written to settings - after wake up the settings subsystem tries to decode the GATT CCC and crashes.

Problem occurs in `subsys/bluetooth/host/gatt.c` in `bt_gatt_store_ccc()` function when it uses `settings_str_from_bytes()` from `subsys/settings/src/settings.c`.

In the log output it's visible that settings subsystems stores BT ID, then BT keys and then tries to store BT CCC which has not been modified as some invalid base64 value.

**To Reproduce**
Steps to reproduce the behavior:
1. Use `samples/bluetooth/peripheral_hids` sample
2. Build for `nrf52_pca10040` or `nrf52840_pca10056`
3. Erase flash `nrfjprog -e` and flash the sample
4. From central - nRF Connect (Windows) connect to peripheral.
5. Pair
6. Perform discovery on HID service - e.g. open its tab in nRF Connect
7. Do not subscribe to any notifications (do not touch CCCD).
7. Disconnect from peripheral
8. Reset board or go to sleep + wake up 
9. Assert:
```
ASSERTION FAIL [rc == 0] @ /home/fi/ncs/zephyr/subsys/settings/src/settings_store.c:50:
        set-value operation failure
```

**Expected behavior**
Value written to flash should be possible to decode or nothing should be written.

**Screenshots or console output**
```
***** Booting Zephyr OS zephyr-v1.13.0-2068-g3984361 *****
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006f1c handle 0x0001 uuid 2a05 perm 0x01
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006f30 handle 0x0002 uuid 2a05 perm 0x01
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006f44 handle 0x0003 uuid 2a05 perm 0x01
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006f58 handle 0x0004 uuid 2a05 perm 0x01
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006f6c handle 0x0005 uuid 2a05 perm 0x01
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006f80 handle 0x0006 uuid 2a05 perm 0x01
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006f94 handle 0x0007 uuid 2a05 perm 0x01
[00:00:00.000,000] <dbg> bt_gatt.gatt_register: attr 0x20006fb4 handle 0x0008 uuid 2a05 perm 0x01
[00:00:00.000,030] <dbg> bt_gatt.gatt_register: attr 0x20006fc8 handle 0x0009 uuid 2a05 perm 0x01
[00:00:00.000,030] <dbg> bt_gatt.gatt_register: attr 0x20006fdc handle 0x000a uuid 2a05 perm 0x00
[00:00:00.473,083] <dbg> bt_gatt.gatt_register: attr 0x20006ff0 handle 0x000b uuid 2a50 perm 0x03
[00:00:00.473,114] <dbg> bt_gatt.gatt_register: attr 0x20006c1c handle 0x000c uuid 2a50 perm 0x01
[00:00:00.473,144] <dbg> bt_gatt.gatt_register: attr 0x20006c30 handle 0x000d uuid 2a50 perm 0x01
[00:00:00.473,144] <dbg> bt_gatt.gatt_register: attr 0x20006c44 handle 0x000e uuid 2a50 perm 0x01
[00:00:00.473,175] <dbg> bt_gatt.gatt_register: attr 0x20006c58 handle 0x000f uuid 2a50 perm 0x01
[00:00:00.473,205] <dbg> bt_gatt.gatt_register: attr 0x20006c6c handle 0x0010 uuid 2a50 perm 0x01
[00:00:00.473,205] <dbg> bt_gatt.gatt_register: attr 0x20006c80 handle 0x0011 uuid 2a50 perm 0x01
[00:00:00.473,236] <dbg> bt_gatt.gatt_register: attr 0x20006c94 handle 0x0012 uuid 2a50 perm 0x01
[00:00:00.473,266] <dbg> bt_settings.bt_settings_init:
[00:00:00.475,311] <inf> bt_hci_core.hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.475,311] <inf> bt_hci_core.hci_vs_init: HW Variant: nRF52x (0x0002)
[00:00:00.475,341] <inf> bt_hci_core.hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 1.13 Build
Bluetooth initialized

[00:00:00.501,739] <dbg> bt_gatt.gatt_register: attr 0x20006b5c handle 0x0014 uuid 2a4c perm 0x01
[00:00:00.501,770] <dbg> bt_gatt.gatt_register: attr 0x20006b70 handle 0x0015 uuid 2a4c perm 0x01
[00:00:00.501,770] <dbg> bt_gatt.gatt_register: attr 0x20006b84 handle 0x0016 uuid 2a4c perm 0x03
[00:00:00.501,800] <dbg> bt_gatt.update_range: start 0x000c end 0x0012 new_start 0x0013 new_end 0x0016
[00:00:00.501,831] <dbg> bt_gatt.gatt_register: attr 0x200069f8 handle 0x0017 uuid 2a4c perm 0x01
[00:00:00.501,861] <dbg> bt_gatt.gatt_register: attr 0x20006a0c handle 0x0018 uuid 2a4c perm 0x01
[00:00:00.501,861] <dbg> bt_gatt.gatt_register: attr 0x20006a20 handle 0x0019 uuid 2a4c perm 0x01
[00:00:00.501,892] <dbg> bt_gatt.gatt_register: attr 0x20006a34 handle 0x001a uuid 2a4c perm 0x01
[00:00:00.501,922] <dbg> bt_gatt.gatt_register: attr 0x20006a48 handle 0x001b uuid 2a4c perm 0x01
[00:00:00.501,922] <dbg> bt_gatt.gatt_register: attr 0x20006a5c handle 0x001c uuid 2a4c perm 0x01
[00:00:00.501,953] <dbg> bt_gatt.gatt_register: attr 0x20006a70 handle 0x001d uuid 2a4c perm 0x10
[00:00:00.501,983] <dbg> bt_gatt.gatt_register: attr 0x20006a84 handle 0x001e uuid 2a4c perm 0x03
[00:00:00.501,983] <dbg> bt_gatt.gatt_register: attr 0x20006a98 handle 0x001f uuid 2a4c perm 0x01
[00:00:00.502,014] <dbg> bt_gatt.gatt_register: attr 0x20006aac handle 0x0020 uuid 2a4c perm 0x01
[00:00:00.502,044] <dbg> bt_gatt.gatt_register: attr 0x20006ac0 handle 0x0021 uuid 2a4c perm 0x02
settings_commit(): name:X
[... 201 lines cut ...]

found handler bt

argc 1 argv[0] id argv[1] (null) val AdA6QU7i/Q==

settings_set_value(): name: bt/keys/de08c8bccdfa1, val_str: EBEgAAAAAAAAAAAAAACc9NtfTAA7gl3aLqyVgiI0AAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA

found handler bt

argc 2 argv[0] keys argv[1] de08c8bccdfa1 val EBEgAAAAAAAAAAAAAACc9NtfTAA7gl3aLqyVgiI0AAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA

settings_set_value(): name: bt/ccc/de08c8bccdfa1, val_str: ��
found handler bt

argc 2 argv[0] ccc argv[1] de08c8bccdfa1 val ��
ASSERTION FAIL [rc == 0] @ /home/fi/ncs/zephyr/subsys/settings/src/settings_store.c:50:
        set-value operation failure
```

**Build environment (please complete the following information):**
 - OS: (Ubuntu 16.04 LTS)
 - Toolchain (e.g Zephyr SDK 0.9.5)
 - Commit SHA: g3984361

Qbicz, 2018-11-21:
@JoeHut is it similar to what you have seen?

Qbicz, 2018-11-21:
@nvlsianpu @Vudentz @jhedberg

Olivier-ProGlove, 2018-11-21:
You need to set `LOG_BT_SETTINGS_CCC_STORE_ON_WRITE=1`. But yes that is the same issue as we have.

Olivier-ProGlove, 2018-11-21:
Be aware of this issue I am working on: https://github.com/zephyrproject-rtos/zephyr/issues/11409#issuecomment-440630063

Qbicz, 2018-11-21:
@Olivier-ProGlove Thanks for the link. `BT_SETTINGS_CCC_STORE_ON_WRITE` fixes the problem in most cases, but not when you perform the discovery but don't subscribe to any notification. There is no CCC write then and this bug reveals.

Olivier-ProGlove, 2018-11-21:
You are right. It is also true we saw this issue even after enabling `LOG_BT_SETTINGS_CCC_STORE_ON_WRITE=1` but much less frequently.

PR #11576 (2018-11-21): Bluetooth: GATT: Fix not resetting CCC storage
If there are no CCC to be stored the value should be set to NULL so it
is properly cleared otherwise calling settings_str_from_bytes will leave
str uninitialized which may cause a crash when attempting to load the
value.

Fixes #11564
commit 046e1ab9 (2018-11-21): Bluetooth: GATT: Fix not resetting CCC storage

If there are no CCC to be stored the value should be set to NULL so it
is properly cleared otherwise calling settings_str_from_bytes will leave
str uninitialized which may cause a crash when attempting to load the
value.

Fixes #11564
closed 2018-11-21 by jhedberg