zephyrproject-rtos/zephyr #11565: gpio_callback ambiguity in union pin vs pin_mask

pabigot, 2018-11-21:
I notice that the gpio_callback structure has a union that's documented as identifying the pins that trigger a callback, expressed as either a single ordinal (`pin`) or a mask of pins (`pin_mask`).  The union documentation in the header [explicitly states](https://github.com/zephyrproject-rtos/zephyr/blob/39843617cbe6f62b195af152de3416a1e4f47399/include/gpio.h#L80) that the application may change the pins associated with the callback at any time.  (Oddly, documentation for members of the struct [does not appear](https://docs.zephyrproject.org/latest/api/io_interfaces.html#structgpio__callback) to be generated.)

There's nothing in the structure that clearly indicates which union member is valid.

As best I can tell, the gpio_utils.h assumes that the `pin_mask` field is the one used; this is also the one initialized by `gpio_init_callback()`.

However, gpio_intel_apl [references](https://github.com/zephyrproject-rtos/zephyr/blob/39843617cbe6f62b195af152de3416a1e4f47399/drivers/gpio/gpio_intel_apl.c#L164) the `pin` element.

Something's hinky here.  Probably the union should be replaced by the `pin_mask` field, any resulting compilation errors fixed, and the suggestion that callbacks can be controlled by pin ordinals removed from the documentation.

pabigot, 2018-11-21:
Thanks; that explains it.  Issue was introduced in PR #10085.  Looping in @dcpleung.

Nonetheless, a field that's public and documented as under application control (since the application owns the structure) now has a driver-specific interpretation.  That's a problem that should be resolved before it bites somebody.  Somebody new might write an application that sets the callback pin by using an ordinal, which will fail on every platform except up_squared.

tbursztyka, 2018-11-21:
yes, that might create an issue in application indeed, mostly a portability one since setting up the cb will not be the same on that controller than other ones.

dcpleung, 2019-01-04:
Sorry for the late reply. I was on vacation. :)

I have given some thoughts about this. I think expanding the APIs to support both pin bitmasks and ordinal pin numbers would lessen the confusion. The GPIO drivers can decide which one to support/implement (or both). As for callback, it is always possible to tag each callback to specify whether the value is a pin mask or ordinal number. However, it would make the ISRs more complicated. I think the drivers should only support one value type in callback.

pabigot, 2019-01-04:
Subordinate devices (I2C, SPI) that have auxiliary GPIOs like reset signals must be able to manipulate those GPIOs in driver code without knowing whether a particular SoC GPIO driver uses mask or ordinal pin specifiers.  How would you support that?

Supporting two incompatible GPIO APIs seems like a far more complex and impactful approach than simply having up-squared split the 40 GPIO pins across two devices.

dcpleung, 2019-01-07:
The Apollo Lake SoC has 245 exposed GPIO pins through the controllers. Given how those pins are arranged in hardware, we will need 10 devices in Zephyr to cover all of these.

I agree that maintaining 2 sets of GPIO APIs are problematic, especially with the need for other drivers to manipulate the GPIOs.

carlescufi, 2019-01-15:
@dcpleung We just reviewed this in the API meeting and we don't deem it acceptable to have such ambiguity in the API. Users have no clue whether the param will be interpreted as mask or pin number. Other SoCs already deal with upwards of 6 GPIO devices, could we revert this and use 10 devices as suggested?

carlescufi, 2019-01-22:
@dcpleung I see you have a branch ready, will you send a PR to fix this?

dcpleung, 2019-01-22:
@carlescufi I have not tested it on actual hardware yet. I will submit a PR once I have tested it on hardware.

galak, 2019-01-29:
> @carlescufi I have not tested it on actual hardware yet. I will submit a PR once I have tested it on hardware.

any update?

dcpleung, 2019-01-29:
Finally found a board, and am testing it now. If everything goes well, I will send a PR later today.

galak, 2019-02-05:
Attempted fix in PR #12850

commit 5b133101 (2019-02-06): Revert "gpio: enable callback to specify pin in addition to pin_mask"

This reverts commit eb6ea28649f6364f17642835af0aa7706c37a10b.

Reverting this to avoid confusion in using the gpio callbacks.

Fixes #11565
closed 2019-02-06 by nashif