zephyrproject-rtos/zephyr #11569: Convert remaining Kconfig symbols of devices available in dts (SPI, I2C)

erwango, 2018-11-21:
Following #11180, some CONFIG_  symbols of devices already converted to DTS remains in dts_fixup.h files:
CONFIG_I2C_X_NAME
CONFIG_SPI_X_IRQ_PRI

There are other devices impacted, like ADC, WDT... but they might not all be available in DTS.
This is different for I2C and SPI as they are available in dts and the other properties are already converted to DT_ namespace.

It apprears theyare still using CONFIG_ because they are defined in Kconfig files.
So netx steps is to remove these symbols from Kconfig and changes definitins from CONFIG_ to DT_

carlescufi, 2018-11-21:
I think this was intentionally left out as `CONFIG_` because they still have the option to use Kconfig for platforms that do not support DT. I will let @galak confirm this.

galak, 2018-11-21:
> I think this was intentionally left out as `CONFIG_` because they still have the option to use Kconfig for platforms that do not support DT. I will let @galak confirm this.

The SPI one needs fixing, need to look at CONFIG_I2C_X_NAME

erwango, 2018-11-21:
What seemed strange to me is that other properties of the same devices are converted.

galak, 2018-11-21:
> What seemed strange to me is that other properties of the same devices are converted.

Not sure why the SPI irq priority wasn't updated.  The CONFIG_I2C_X_NAME case is because I2C slave drivers need to all be updated before we can remove it.

erwango, 2018-11-21:
> [...]The CONFIG_I2C_X_NAME case is because I2C slave drivers need to all be updated before we can remove it.

Ok, got it, txs

erwango, 2018-11-22:
Closing the issue

closed 2018-11-22 by erwango