zephyrproject-rtos/zephyr #11594: Cleanup GNUisms to make the code standards compliant

TTornblom, 2018-11-22:
**Is your feature request related to a problem? Please describe.**
There is work in progress to allow Zephyr to use non-gcc based toolchains. Some of these do not allow many of the non-standard extensions that seems to be commonly used in the Zephyr code.

I started trying to build the **hello_world** sample on **IAR Embedded Workbench for ARM** and quickly ran into at least three issues:
1) Empty structs:
```
struct _caller_saved {
    /*
     * Unused for Cortex-M, which automatically saves the necessary
     * registers in its exception stack frame.
     *
     * For Cortex-A, this may be:
     *
     * u32_t a1;       r0
     * u32_t a2;       r1
     * u32_t a3;       r2
     * u32_t a4;       r3
     * u32_t ip;       r12
     * u32_t lr;       r14
     * u32_t pc;       r15
     * u32_t xpsr;
     */
};
```

2) Case ranges:
```
           switch (*fmt) {

            case '1' ... '9':

            }
```

3) Zero length arrays:
```
extern struct log_source_const_data __log_const_start[0];
extern struct log_source_const_data __log_const_end[0];
```

I'm confident that there will be other issues as well.

**Describe the solution you'd like**
I suggest cleaning up the code to make it standards compliant (the IAR tools support C89 and C11).

**Describe alternatives you've considered**
Adding the used extensions to the new toolchains is not really a viable option.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

d3zd3z, 2019-12-20:
`_caller_saved` has been removed in #15541

d3zd3z, 2019-12-20:
Master (as of this comment) has the following `alloca` calls:
```
include/sys/rb.h:                       alloca((tree)->max_depth * sizeof(struct rbnode *)), \
include/sys/rb.h:       .is_left = (char *)alloca((tree)->max_depth * sizeof(char)),            \
lib/os/rb.c: * alloca().  The current node is found in stack[top] (and its parent
subsys/logging/log_output.c:    u32_t *args = alloca(sizeof(u32_t)*nargs);
subsys/logging/log_output_syst.c:       u32_t *args = alloca(sizeof(u32_t)*nargs);
tests/subsys/logging/log_immediate/src/log_immediate_test.c:    u8_t *buf = alloca(buf_len);
```

mped-oticon, 2019-12-20:
And if I recall correctly, the red black trees can be avoided if using dumb scheduling.

d3zd3z, 2019-12-20:
The ones found in my above comment should be easily replaced with dynamic arrays. Whether that is the desired solution is another question, but the generated code should be equivalent.

mped-oticon, 2019-12-20:
@d3zd3z if by dynamic arrays you mean VLAs, then that used to be the case; rb changed away from VLA to alloca because of poor VLA support in compilers (alloca was at least an explicit implementable function with same behavior).

Finally, if rb's alloca remains an issue, it should be fairly easy to add a kconfig replacing alloca calls with statically dimensioned arrays, upper-bounded by rb's log-height.

d3zd3z, 2019-12-21:
@mped-oticon I'm kind of surprised that VLA would have less support than alloca. VLA is part of C99, alloca never has been. I'm curious how alloca would be implemented in a portable way, without compiler support for it.

General consensus (stack overflow search for example) seems to suggest moving away from alloca to VLAs, since that is defined by the language. But, we may have to do something else for the cases where neither are available.

andrewboie, 2019-12-23:
> extern struct log_source_const_data __log_const_start[0];
> extern struct log_source_const_data __log_const_end[0];

this is trivial, just remove the zero in favor of empty brackets, it's just a linker symbol anyway

sslupsky, 2020-04-07:
How about bringing things up to date so we can use c18?

c99 seems way outdated for an RTOS in 2020?

d3zd3z, 2020-04-10:
> c99 seems way outdated for an RTOS in 2020?

People still complain from time to time when C99 features are used. Just try declaring a variable inside a block of code.

manoj153, 2021-10-12:
> Hello @manoj153, The effort of supporting Zephyr with IAR is stalled, unfortunately. I hope I'll be able to work on it again soon.
@TTornblom 
Possible for any of us can help to help to expedite?

gregshue, 2022-02-23:
Standards compliance means limiting globally scoped symbols to 31 significant characters.  Is that within the scope of this ticket?

TTornblom, 2022-02-23:
> Standards compliance means limiting globally scoped symbols to 31 significant characters. Is that within the scope of this ticket?

I have not seen this as a problem with Zephyr, and the IAR compiler does not seem to have a problem with this.

gregshue, 2022-02-23:
Interesting, but independent of whether or not our code is compliant with the C99 standard.

TTornblom, 2022-02-23:
My main issue has been with gcc extensions that aren't easily worked around using standard C, like typeof, alloca, case ranges (...), variable length arrays, etc. If symbol length would have been an issue they could easily have been shortened to make them work.

gregshue, 2022-02-23:
Understood, but this issue seeks standards compliance, not just portability. 😉

d3zd3z, 2022-02-25:
I don't think we can reasonably say we comply with identifier significant character restrictions without some kind of tool that checks for naming clashes due to this.

But, I will ask, are you aware of any compiler that only checks the first 31 characters of an identifier? The C standard encourages tools to not have restrictions like this. Implementing a check for this could be a good bit of work, since I haven't been able to find any options in either clang or gcc to check for it.

Mostly, I would expect long identifiers like this to result from macros that build names. We do this quite a bit, and it would be a good idea to make sure that the generated names do differ earlier on. But, again without a tool to check this, it would be hard to state whether we comply or not.

Practically, the best we can do is say that we work with a certain list of compilers, and deal with the things that the tools we have are able to catch (e.g. `--pedantic`).

gregshue, 2022-02-27:
> Mostly, I would expect long identifiers like this to result from macros that build names.

That is one source.  Others are:

- testcase names (which should be descriptive).  Mine have already pushed the 64 significant character limit.
- any prefixes analogous to C++ namespaces (which be nested and hierarchical).
- avoiding variations in abbreviations (e.g. using "manufacturing" rather than "mfg", or "internationalization" instead of "i18n").

> Practically, the best we can do ...

I think there is a better practical solution. Users need the codebase to work with compilers of their choice, and contributors need to know the constraints that need to be applied to their designs and implementations. 
 We could say Zephyr requires toolchains that support N significant characters (e.g. 255) in linker identifiers, and create two identifiers in the kernel that depend on it and trigger a build error if the toolchain does not support it.  That would ensure implementers can design around a larger number of significant characters.

Thoughts?

stephanosio, 2022-11-14:
To be tracked in https://github.com/zephyrproject-rtos/zephyr/issues/52226

closed 2022-11-14 by stephanosio