zephyrproject-rtos/zephyr #11605: Convert lis2dh to use dts

galak, 2018-11-22:
Add DTS bindings for lis2dh and update driver to get info like SPI bus name, GPIOs, etc from DTS generation.

galak, 2018-11-22:
@avisconti Can you take a look

avisconti, 2018-11-22:
ok

avisconti, 2018-11-22:
I'm looking at other sensors that already support dts. It seems that still is not possible in zephyr dts subsystem to discriminate between I2C or SPI, and still requires Kconfig to do so. Correct?

pabigot, 2018-11-23:
From my experience I think the "natural" DT way to do it requires two binding variants: one that inherits from `spi-device` and one from `i2c-device`.  Then the DT fixup would have to do something like:

```
#ifdef DT_NORDIC_NRF_I2C_40003000_BOSCH_BME280_45_LABEL
#define DT_BME280_I2C_NAME DT_NORDIC_NRF_I2C_40003000_BOSCH_BME280_45_LABEL
...
#elif defined(DT_NORDIC_NRF_SPI_40003000_BOSCH_BME280_45_LABEL)
#define DT_BME280_SPI_NAME DT_NORDIC_NRF_SPI_40003000_BOSCH_BME280_45_LABEL
...
```
and the driver could switch off the defined symbol without involving Kconfig.

The latter part is straightforward.  I have no idea how to do the yaml to describe two ways of connecting the same sensor.

avisconti, 2018-11-23:
@pabigot 
Yes, this is the way I'm doing and all the sensors are implemented.
So, I created this twio yaml descriptions:
dts/bindings/sensor/st,lis2dh.yaml
dts/bindings/sensor/st,lis2dh-spi.yaml

But this requires that in config file you configure SPI or I2C bus type.

I'm not so familiar with the yaml way to implement dts, but in case of linux dts which bus a device is attached to is an info actually extracted from dts (I think). So I was wondering if zephyr can do the same or not in the future.

avisconti, 2018-11-23:
@pabigot 
OK, I see now what you are saying about dts_fixup.

pabigot, 2018-11-23:
In Linux I believe device information is extracted from dtb (the binary form compiled from the DT source) at runtime.  Zephyr seems to extract the DT information from dts and use it only at compile-time, through dts_fixup.

My solution to a question @galak asked in #11545 about how to support multiple devices would be to evolve Zephyr infrastructure where drivers did configure themselves at boot time from dtb, as encapsulating the hardware connections in a data structure that's part of the device object is a well-defined way to do multiple instances.  I also don't think that's likely to be supported anytime soon as it's a big architectural change.

avisconti, 2018-11-23:
Wouldn't it be possible to modify the scripts to set BUS_TYPE based on which  parent bus is included?

pabigot, 2018-11-23:
If you're asking me I have absolutely no idea.  The closest I've gotten to what Zephyr does with the content of device tree source is create and modify yaml files.  Anything between that and dts_fixup is, at this point, magic to me.

closed 2019-01-24 by galak