zephyrproject-rtos/zephyr #11617: net: ipv4: udp: broadcast delivery not supported

rmstoi, 2018-11-23:
Zephyr drops UDP packet with broadcast destination IP and unicast MAC. According to RFC 1122 4.1.1 this scenarion should be supported by UDP: "... to use communications services (e.g., multicast or broadcast delivery) not available from TCP."

[ipv4-udp-bcast-dst.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2610099/ipv4-udp-bcast-dst.pcap.gz)

jukkar, 2019-01-16:
Works for me, perhaps this was already fixed or there was some issue during testing. Anyway, attached is a capture of an UDP packet with broadcast destination address and it was handled by zephyr just fine.

[ipv4-udp-bcast-dst-example.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/2763936/ipv4-udp-bcast-dst-example.pcapng.gz)

rmstoi, 2019-05-15:
New pcap, with correct Zephyr MAC address.

[ipv4-udp-bcast-dst2.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/3182048/ipv4-udp-bcast-dst2.pcap.gz)

PR #16172 (2019-05-20): net: ipv4: Allow UDP packets with broadcast dst address
Make sure we are able to receive UDP packets with broadcast
destination address.

Fixes #11617
commit 7798df4b (2019-05-20): net: ipv4: Allow UDP packets with broadcast dst address

Make sure we are able to receive UDP packets with broadcast
destination address. If CONFIG_NET_IPV4_ACCEPT_ZERO_BROADCAST
is set, then check here also non-standard broadcast address
that is described in RFC 1122 chapter 3.3.6.

Fixes #11617
commit 28485586 (2019-05-20): net: ipv4: Allow UDP packets with broadcast dst address

Make sure we are able to receive UDP packets with broadcast
destination address. If CONFIG_NET_IPV4_ACCEPT_ZERO_BROADCAST
is set, then check here also non-standard broadcast address
that is described in RFC 1122 chapter 3.3.6.

Fixes #11617
commit b800737f (2019-05-20): net: ipv4: Allow UDP packets with broadcast dst address

Make sure we are able to receive UDP packets with broadcast
destination address. If CONFIG_NET_IPV4_ACCEPT_ZERO_BROADCAST
is set, then check here also non-standard broadcast address
that is described in RFC 1122 chapter 3.3.6.

Fixes #11617
closed 2019-05-20 by jukkar