zephyrproject-rtos/zephyr #11618: net: icmpv4: Zephyr drops valid echo request

rmstoi, 2018-11-23:
Zephyr drops echo request with valid checksum. Zephyr must respond to valid echo request.

[icmpv4-valid-chksum2.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2610165/icmpv4-valid-chksum2.pcap.gz)
[icmpv4-valid-chksum1.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2610166/icmpv4-valid-chksum1.pcap.gz)

jukkar, 2019-01-16:
Works for me, perhaps a glitch during the test?

rmstoi, 2019-01-17:
That is weird, cause when I test these packets are still dropped:

```
[00:08:50.760,000] <dbg> net_ipv4.net_ipv4_process_pkt: (0x566515d0): DROP: invalid chksum
[00:08:50.760,000] <dbg> net_core.processing_data: (0x566515d0): Dropping pkt 0x56639d24

```

tbursztyka, 2019-01-17:
Here it's the IPv4 chksum which is invalid, not the ICMPv4 one.

rmstoi, 2019-03-11:
@tbursztyka I tried PR #14200, now echo server replies to ping, but the response has wrong id and  seq numbers. So wireshark marks request as "No response seen"

[icmpv4-valid-chksum1-pr14200.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2951388/icmpv4-valid-chksum1-pr14200.pcap.gz)

tbursztyka, 2019-03-11:
@rmstoi previous echo reply build was cloning the original packet, as the un-existing ipv4 options made it assume ipv4 hdr is always same which is not. I modify the echo reply build to it only copies the payload from original packet, but builts ipv4/icmpv4 hdr properly. Should be fine now. (I quickly tried and no complaints from ws)

rmstoi, 2019-03-11:
@tbursztyka works like charm. Thanks!

commit a567bcb7 (2019-03-12): net/utils: Use the right IPv4 header length for checksum calculation

IPv4 header length might be bigger than struct net_ipv4_hdr if there are
options appended to it.

Fixes #11618
closed 2019-03-12 by andrewboie