zephyrproject-rtos/zephyr #11625: Problem with printk and logger used with shell

MarekPieta, 2018-11-23:
**Describe the bug**

Bug occurs when: 

- using shell on RTT backend,
- calling LOG_ERR and printk functions (printk is processed by logger),
- log internal thread has low priority (for example: 2).

On logger thread priority=2 some messages do not appear (it shows only output from printk and not fom SYS_ERR). Everything is displayed properly when logger thread has priority -2 (default for sample). In general, in many cases logger is used with low priority, so it may cause problems.

**To Reproduce**
Using subsys/shell/shell_module sample. Changed prj.config to:
`CONFIG_PRINTK=y

CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_USE_SEGGER_RTT=y

CONFIG_LOG_BACKEND_UART=n
CONFIG_UART_CONSOLE=n

CONFIG_LOG_BACKEND_RTT=n
CONFIG_LOG_PROCESS_THREAD_PRIO=2
CONFIG_LOG_PRINTK=y

CONFIG_SHELL_BACKEND_SERIAL=n
CONFIG_SHELL_BACKEND_RTT=y
CONFIG_KERNEL_SHELL=y
CONFIG_THREAD_MONITOR=y
CONFIG_INIT_STACKS=y
CONFIG_BOOT_BANNER=n
CONFIG_THREAD_NAME=y
CONFIG_DEVICE_SHELL=y
CONFIG_GPIO_SHELL=y
`
To main.c added another thread:
`
#define THREAD_STACK_SIZE 500
#define THREAD_PRIORITY 5

void thread_fn(void)
{
	while(true){
		LOG_ERR("Thread Log");
		printk("Thread Printk \n");
		k_sleep(1000);
	}

}

K_THREAD_DEFINE(test_thread_id, THREAD_STACK_SIZE,
                thread_fn, NULL, NULL, NULL,
                THREAD_PRIORITY, 0, K_NO_WAIT);
`
and calls to functions: printk, SYS_ERR in main function
`void main(void)
{
	shell_prompt_change(shell_backend_rtt_get_ptr(),
			    "test>");
	while(true){
		LOG_ERR("Main log 1");
		printk("Main printk 1 \n");
		LOG_ERR("Main log 2");
		printk("Main printk 2 \n");
		printk("Main printk 3 \n");
		k_sleep(1000);
	}
}
`

Reproduced on: nrf52_pca10040.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 16.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.9.5
 - Commit SHA: 92f7c7c3b511282b5fe4550dd217b5b30fef2057

carlescufi, 2018-11-23:
CC @pkral78

nordic-krch, 2019-01-30:
@MarekPieta, i think this is fixed. Could you confirm?

pdunaj, 2019-01-30:
seems working now after the last update

closed 2019-01-30 by nordic-krch